opentx/opentx #3150: PREC1 and PREC2 formatting parameters not functioning correct.

lshems, 2015-12-14:
the "PREC1" parameter used to format numbers being printed with1 decimal is working differently. It used to just place a decimal point before the last whole number digit. Now it seems to just give the 1/10 value in the output after the digit, so it is really only a "format" parameter.

value 67, OLD 'PREC1' format output: 6.7, new 'PREC1' format output: 67.0

also, there seem to be some rounding errors. 8.2 is formatted as 8.19 with 'PREC2'

projectkk2glider, 2015-12-14:
Confirmed:
```lua
local function run(event)
  lcd.clear()
  lcd.drawNumber(10, 10, 1234, LEFT)
  lcd.drawNumber(10, 20, 1234, PREC1+LEFT)
  lcd.drawNumber(10, 30, 1234, PREC2+LEFT)
  return 0
end

return { run=run }
```
produces:
[image]

>also, there seem to be some rounding errors. 8.2 is formatted as 8.19 with 'PREC2'

That is Lua rounding error when the floating point number is converted to integer. We can't do much about it.

projectkk2glider, 2015-12-14:
Looks like this was a deliberate change done in bab4168ac435c7d5d997648dcd4219be57dcb915. So the question for @bsongis:
 * do we change the documentation to reflect the new funcitonality
 * or we change back the functionality the way it was in 2.0?

I vote for the first option.

bsongis, 2015-12-14:
I also prefer the first option

bsongis, 2015-12-14:
The Lua rounding error should be reported in a different issue, perhaps we can do something!

lshems, 2015-12-15:
the rounding doesn't occur when just shifting the decimal. so for that reason I would prefer the second option. It is only a formatting issue, and num2text does it correct.

bsongis, 2015-12-15:
Well I have nothing against the 2nd solution, it would be consistent with the firmware itself. Just a little bit more difficult to explain.

projectkk2glider, 2015-12-15:
Yes, the second option would be consistent. For a more elaborate string formatting one can use the Lua string library now.

closed 2015-12-16 by bsongis

lshems, 2015-12-16:
Bertrand,

I am new to all this github stuff.

The issue about PREC1 and PREC2, do i see it correct if the original
behaviour is being restored, that is 30 PREC1 is dislayed as 3.0 ??

How can I see in which nightly it will be updated so I can test??

Thanks for the reply.

Guido

Met vriendelijke groet

Guido ter Horst
+32 499 708000

bsongis, 2015-12-16:
15 hours, it will be in the nightly builds tomorrow morning!

lshems, 2015-12-17:
Perfect.

If this is fixed, I have to do some final testing, and then I can release my updated F3K toolset.compatible with 2.0 and 2.1.

:)

astrohome, 2016-01-09:
Any news concerning this bug?

projectkk2glider, 2016-01-09:
It's fixed in 2.1.7

astrohome, 2016-01-09:
@projectkk2glider my bad - I haven't seen that it's closed already.

commit 8942115f (2015-12-16): Fixes #3150
commit d4c02718 (2015-12-16): Fixes #3150