opentx/opentx #3155: Nightly builds failure

driedeker, 2015-12-16:
Not sure if its just me but the nightly builds fail to down load in companion and strait from the web. probably over the last 4 to 5 days.
error i get is server closed.
Terry McKinley

kilrah, 2015-12-17:
Can't see a problem here. 
But surely you might want to add some relevant details like what radio type and options you're trying to download...

driedeker, 2015-12-17:
I am on windows 7 ult and using taranus, its companion nightly builds i am on version 216N573
which did download.
When i startup companion it tells me a new version is available did i want to download ans after saying yes it gets to 98-99% and tells me  with a message box saying the download has failed connection closed.
if i go direct to the download with a browser of say 216N575 it failes yet a second later i got 216n574 no problem.
Very strange its not all the builds that fail and I i have no other problems with downloading the firmware updats.

driedeker, 2015-12-17:
216N575 downloaded in chrome but failed file check when run yet version 216n574 downloaded and installed ok.

kilrah, 2015-12-17:
I just downloaded companions from N572 to N574, they all run and install correctly. Then launched N574 and checked for updates, it found N575 as it should, and that downloaded and installed correctly.
Must be an issue on your end with your computer or internet connection...

closed 2015-12-17 by kilrah