opentx/opentx #3156: servo offset of 6.6 stored as 6.5 (companion and radio?)

lshems, 2015-12-16:
When setting in companion the servo offset to a value of 6.6, it is stored as 6.5.
You can change it thorugh the radio setup screen, which will adjust the underlying value.

Use this script to test, Set servo offset on channel 0 to 6.6 and go to the telemetry screen to check values.

This must/is related to the PREC1 PREC2 issue #3150, since the model.drawNumber function with PREC1 does improper roundoff, also demonstrated in the attached script.

Replacing the offset value by 66 as hardcoded integer value produce the same erroneous result

Code to be run as telemetry script:

local function run()
  lcd.clear()
  local x=10
  local y=1
  
  lcd.drawText(x+40,y,"drawNumber ch1 offset")    
  lcd.drawNumber(x,y,model.getOutput(0)["offset"],LEFT)
    
  y=y+10
  lcd.drawText(x+40,y,"drawNumber (ch1 offset/10)")
  lcd.drawNumber(x,y,model.getOutput(0)["offset"]/10,LEFT)
  
  y=y+10
  lcd.drawText(x+40,y,"drawText (ch1 offset/10)")
  lcd.drawText(x,y,model.getOutput(0)["offset"]/10)
  
  y=y+10
  lcd.drawText(x+40,y,"drawNum(ch1 offset/10) PREC1")
  lcd.drawNumber(x,y,model.getOutput(0)["offset"]/10,PREC1+LEFT)
  
  y=y+10
  lcd.drawText(x+40,y,"drawNumber (66 / 10) PREC1")
  lcd.drawNumber(x,y,66/10,PREC1+LEFT)
   
  y=y+10
  lcd.drawText(x+40,y,"drawTEXT (66 / 10)")
  lcd.drawText(x,y,66/10)
end

return {run=run}

kilrah, 2015-12-16:
I believe it's unrelated, and you're rather seeing the 1000/1024 ratio that exists between setting and servo output.

lshems, 2015-12-16:
Nop, i've checked the 1000/1024 ratio, and for outputs, which are noted as -100% - 100% the model.getinput() will return the % value * 10, and not the % value * 10.24.

See screenshots:
[image]
[image]

lshems, 2015-12-16:
I think the culprit is in the way you have ordered things. If you first multiple the argument with the prec factor(10 or 100) and then do the conversion to integer and then proceed in processing the result with the decimal formatting flag, all is fine.

projectkk2glider, 2015-12-17:
It is a problem with drawNumber and PREC which was already fixed in  #3150. Please test with nightly  build #575 or later.

lshems, 2015-12-17:
Thanks :)

Steep learning curve here.

LOL

lshems, 2015-12-17:
Tested, unfortunately, the companion still doesn't save some values correct. 6.6 is saved as 6.5.

The simulator works correct and stores correct.

projectkk2glider, 2015-12-18:
The remaining problem/solution is probably the same as https://github.com/opentx/opentx/issues/3117#issuecomment-163155722

lshems, 2015-12-18:
> The remaining problem/solution is probably the same as #3117 (comment)

Agree. Not blocking for me, but not pretty either.

Is it on the radio as we'll?

het volgende:

projectkk2glider, 2015-12-26:
>When setting in companion the servo offset to a value of 6.6, it is stored as 6.5.

Similar problem to #3117, which I can not reproduce (again :angry:) Anyway there is a fix done, @lshems please test next the nightly build and report back.

commit a9b3b403 (2015-12-26): Fixes #3156: Outputs Subtrim, Min and Max values rounding problem fixed
commit 117e28db (2015-12-26): Fixes #3156: Outputs Subtrim, Min and Max values rounding problem fixed (ported from master)
closed 2015-12-26 by projectkk2glider