opentx/opentx #3157: LUA modelfuctions feature request making half the model functions obsolete

lshems, 2015-12-16:
Why do we not return the resulting settings on a model.setXXX() call??

This would make the model.getXXXX() obsolete, improve reliability and reduce maintenance.

Use Case:
--I want to set the 'offset' of output '0' to 'newValue', and be sure that 'newValue' will contain the actual value after the 'set' action, even if the call didn't succeed in updating the setting.

COde examples:

--------------------------------------------------------------------------------------
local newValue
---------------------------------------------------------------------------
--code example current conventions
newValue=10
model.setOutput(0,{"offset"=newValue})
newValue=model.getOutput(0)["offset"]
--newValue will be 10

newValue=10000
model.setOutput(0,{"offset"=newValue})
newValue=model.getOutput(0)["offset"]
--newValue will still be 10 after these instructions

----------------------------------------------------------------------------
--proposed 1-line instruction doing the same, if setOutput will return the resulting settings
newValue=model.setOutput(0,{"offset"=newValue})["offset"]

--proposed 1-line instructions equivalent to model.getOutput(0)
newValue=model.setOutput(0)
--or
newValue=model.setOutput(0,nil)
--or
newValue=model.setOutput(0,{"offset"=nil})

projectkk2glider, 2015-12-17:
I've taken the liberty to format your text above. We will need to discuss this a little more before implementing.

First your example:
```lua
newValue=10000
model.setOutput(0,{"offset"=newValue})
newValue=model.getOutput(0)["offset"]
--newValue will still be 10 after these instructions
```
Last line: newValue should be 10000, or I am missing something?

This could actually be easy to implement, every existing set function would function like now (if parameter is given) and also return the current state of set parameter. 

The main concern for me is to keep the OpenTx Lua API  compatible order to keep existing scripts running and not to confuse users. So my proposal:
* add the new functionality to the existing setXXX functions (they will also work when used in the old way)
* we keep the existing getXXX functions for backwards compatibility (perhaps add note in docs that this functions are depreciated and will be removed in some future release...)

kilrah, 2015-12-17:
My understanding is that his example you quoted is "how it works now", "how he wants it to be" comes later.

lshems, 2015-12-17:
two important things here: 

1) I intentionally gave the example with 100000 as newValue. the output cannot be set to this value, so what will happen? I think it should stay as it was without attempting to update it.

2) the return value is not simply the value you sent. It needs to reflect the setting after the (attempt to) update

Why? because now, if I don't check the newvalue to be within allowed rannge before trying to update, I am not sure wheather the setting got updated or not. By returning the result of the set action, I am always sure about the result of my attempt, and never in an undefined state.

Guido

P.S.
I agree with the approach of leaving the set intact, just have it return the result. The get should remain for backward compatibility only, and behind the screens just be a forwarder to the set fuction. This in order to minimise code maintenance and risk of erronuous implemention of two linked commands when code changes are needed for some reason.

lshems, 2015-12-29:
Just for my understanding, how are feature request followed up?? Is there a priority?

kilrah, 2015-12-29:
The priority is "what someone feels like doing, or thinks is important" once things have been discussed and accepted. As you can see from the number of open issues we have much less time/resources than needed to do everything that's suggested or desirable, so some things get picked and done same day while others can stay in the queue for years (at which point someone else, why not you, could pick them up and implement them!)

In this case a consensus hasn't been reached at this point, so it's just dangling there... 

To me:
> Why? because now, if I don't check the newvalue to be within allowed range before trying to update, I am not sure whether the setting got updated or not.

There's a golden rule called [sanitize your inputs](http://xkcd.com/327/) that calls for you precisely doing that, and implies that if you do you can expect the result to be correct.

Think about it - you'll have to do a test anyway, currently you *should* be testing whether the input is sane, and with your change you'd have to check the return value matches what you sent. But on our side we can currently just assign your value without further ado because we know it's valid (you've sanitized it), while with your request we'd have to add a test on our side to validate your input that may or may not fit, i.e. one more test and the related overhead.

We'd be going from:
- Your test -> valid input -> Function call -> result is known good because input was good  

to
- Function call with unknown input -> Our test to check you didn't send garbage -> your test to check you didn't send garbage and cause the call to fail.

Can't see how that would be considered any reasonable. 

So - let's just drop this, and you sanitize your inputs ;)

lshems, 2015-12-29:
Sorry, i don't fully agree about the checking things. What happens if I use
the setValue with a not allowed value? You need to check that anyway, no?

Even if you don't check, as long as I don't cause an error when sending an
insane value, and use the setValue to check what has happened, I'll get the
desired result: a return value of the actual state, changed or not.

So if you would just call the getValue at the end of the setValue call and
return the result, I would be perfectly happy without any extra programming
effort.

?

kilrah, 2015-12-29:
> Sorry, i don't fully agree about the checking things. What happens if I use the setValue with a not allowed value? 

It will break things, so you must make sure not to do it. Just look at the code in the API. 

```      
else if (!strcmp(key, "channelsCount")) {
        module.channelsCount = luaL_checkinteger(L, -1) - 8;
```

We take your value and assign it, period. If you send 135, channelsCount will be set to 135-8 which is not valid at all, and might even cause an overflow, crash, etc.
And getting it would return 

```    lua_pushtableinteger(L, "channelsCount", module.channelsCount + 8);```

i.e. 135 again even if it doesn't mean anything.

We do not check validity. You must do it.

lshems, 2015-12-29:
OK. That's clear. (And dangerous lol)

kilrah, 2015-12-29:
Well if you want to manipulate model data you sure want to be sure of what you do - and the right moment to check that is when you write the script, not when a user is running it.

closed 2015-12-29 by kilrah

lshems, 2015-12-29:
I have range control on all settings ( if I didn't forget anything) and it
is in my standard routines. But still, ...... :)