opentx/opentx #3158: RADIO feature request

lshems, 2015-12-16:
I currently use physical switches as source on inputs, which in turn trigger logical switches.

In this way, all special functions or mixes or whatever is based on this logical switch, can be assigned to another physical switch in just 1 place.

If I do not want a physical switch to control the logical switches linked to it, but I want to assign a fixed value to this input to 'simulate' a fixed switch position, freeing up the physical switch for other functions, I assign the MAX source.

This works fine, with one drawback:
  I need to set the weight to -100% at the ssame time, to simulate the default 'up' switch position.

If we would add another source value DEFAULT, being -100%, or -1024, you could very conveniently replace the physical switch source with the DEFAULT source.

This becomes extremely convenient when using LUA-scripts to allow for changing these assignments. You simply include the DEFAULT source as a possible choice between the physical switch source identifiers, and then you don't have to take care of changing the weight when reassigning sources.

DEFAULT is just a proposed name, perhaps there are better.

If my explanation is not clear, please let me know.

lshems, 2015-12-29:
A lot of people use values to check weather a switch is set or not in the LS screen. When the source MAX would be renamed "DOWN", and a new source with value -100% would be added as "UP", you can use those in your LS settings and it becomes much more understandable.

It would also sort out my needs of allowing people to replace a physical switch with a "fixed" switch position in a logical manner.

jtaylor2, 2015-12-29:
@lshems 
> When the source MAX would be renamed "DOWN", and a new source with value -100% would be added as "UP", you can use those in your LS settings and it becomes much more understandable.

In my opinion renaming MAX to DOWN would be totally confusing and not understandable at all.  First of all, I (and I expect most people) consider up positive and down negative so having UP -100 and DOWN +100 would be counter intuitive.  Consider the typical throttle safe mix line of:

CH3 R MAX (weight -100)  Switch (SF↑)  (Thr-Safe)
Pretty clear the throttle is set to -100 (throttled down)

Now consider:
CH3 R UP  Switch(SF↑}  (Thr-Safe)
Not clear at all that UP sets the throttle channel is throttled down.

Also, in my opinion, changing the name of something that is documented everywhere, and thousands of users understand and use to make someones single use case clearer doesn't make a lot of sense.

lshems, 2015-12-29:
Agree. Do you have any suggestions as alternative to remove a switch
assigned to a AND function on a LS and give it a default switch value?

I have several LS in a sort of standard épée, and use some to actually
switch, and some should just be set or unset, depending on the model.

kilrah, 2015-12-29:
Exact, UP or DOWN can be confusing and misleading. Even with switches, for some UP is "away from them", for others UP is towards them (at least for the shoulder switches, because if you put the radio down on its back then the switch is UP when it's towards the user. 

Then MAX still has to stay there for other arbitrary fixed values anyway.

kilrah, 2015-12-29:
> Do you have any suggestions as alternative to remove a switch assigned to a AND function on a LS and give it a default switch value?

Use an input.

jtaylor2, 2015-12-29:
>Agree. Do you have any suggestions as alternative to remove a switch assigned to a AND function on a LS and give it a default switch value?

No, but I can't understand the use case of having a fixed (default) value  to an AND  function.  If one parameter to a logic function isn't going to change you don't need the function.....just use either the first parameter or the !first parameter.  Perhaps I don't understand the question, but I don't see how removing a switch (physical, logical, or bool) from and AND function and replacing it with a default value would be used.

lshems, 2015-12-29:
LS1 sets of a function on Aup
LS2 sets of a function on Adown

I don't want to use the switch A anymore for these functions, since I have
it always at Aup. Now what I do is assign LS 31 which I forced to true at
the place of Aup, and assign LS32, which is always false, at position of
Adown.

I need 2 LS to do that, because I don't have a fixed value at hand to
assign.

This is not a problem at all when managing everything inside the standard
screens, but if you are using scripts to do that, it becomes akward not
being able to 'disable' a switch without assigning a new function. It
becomes really messy.

kilrah, 2015-12-30:
Why assign different switches??

Just replace LS1 with something like OR SAup !SAup. or other always true condition, and LS2 with some always false condition.
Or even better, propagate the change where it applies. It would be a pain if done manually, but not not from within a script.

Or you could possibly use inputs as I suggested depending on what you're trying to do. Not understanding what you're working on makes it harder to suggest alternatives.

lshems, 2015-12-30:
André,

The use case only involves being able to write proper lua scripts, allowing
to replace some switch (Aup for instance) with a fixed value in the eepe,
without knowing the context or other functions it triggers.

I do it now by creating an input with source Max, and change the source to
the switch and back again. I never use physical switches anywhere else in
my eepe because it is a personal preference, so should be changed in one
place only.

But Max is corresponding to switchdown, so that defaults to the wrong
value. That's why I want a source with an opposite value as well,
referenced by just one identifier.

I think with the introduction of Lua, there are some use cases especially
for supporting easy programming in lua. They are not neccesary when not
using lua.

lshems, 2016-01-07:
Perhaps it's to early for this. I'll program around it.

closed 2016-01-07 by lshems