opentx/opentx #3159: Horus port

bsongis, 2015-12-17:
**Port from master branch**
- [ ] S.PORT Powerbox support
- [ ] Crossfire

**Can be done before we have the full radio in hands:**
- [ ] Implement the CurvesAll screen designed by Martin
- [ ] Model Selection: add the possibility to move models and categories in the list
- [ ] Add the support for jpg and/or png
- [ ] Lua: export all LCD functions to the interpreter
- [ ] Lua: design a test / demo screen to show the possibilities
- [ ] Add a way to override all default main views by custom ones, Lua based

**Can be done after we have the full radio in hands:**
- [ ] ON and OFF sequences
- [ ] Analogs and switches
- [ ] Pulses: PPM / DSM2 / PXX / Crossfire
- [ ] FrSky D / S.PORT telemetry
- [ ] RTC
- [ ] Watch Dog Timer reboot

hrenz, 2015-12-20:
Hy Bertrand
Can be done before we have the full radio in hands:
Language specific characters at companion and simulator
Screens size reduction  to max 1000x750 pixel for normal 15'' Notebooks at simulator
At Log switches the infos for the Inputs: Funktion, V1, V2 AND  .....
Beep Pitch at radios setup at Companion, now only at TX
Spezial funktions Inc / Dec not only +/-1   frees settings +/-1  .....   +/-10 ..... +/-25

kilrah, 2015-12-20:
These are all existing issues and none of them are directly related to or needed for the Horus. I.e. nothing to do in here.

projectkk2glider, 2015-12-20:
One down :smile: , many to go...

bsongis, 2015-12-26:
Yes but in fact no, the adaptation to the Horus GUI is needed

closed 2016-08-01 by bsongis