opentx/opentx #3160: Add commandline options to standalone simulator

projectkk2glider, 2015-12-17:
* radio type
* eeprom file to use (could also be added to the first dialog box)

lshems, 2015-12-17:
Oh yeah, command line options I can use in a windows link.

Then I don't have to select the one radio I have every time:)

GoGoGo

P.S. option in companion to launch simulator, since it allows to change settings etc and have them saved to the taranis.bin file. Some things cannot even be changed in any other way.

bsongis, 2015-12-17:
Yes it's a good idea!

projectkk2glider, 2015-12-17:
Idea is not mine :grin: , but I do like it.

pointu, 2015-12-18:
The questioning is on on the French taranis forum, something like :
"simulator.exe -f=myTaranis.eepe -m=Salto sc=telemetry"

bsongis, 2015-12-18:
I hope @projectkk2glider won't choose this syntax!

pointu, 2015-12-18:
let me know ?

kilrah, 2015-12-18:
Radio type and eeprom file are easy to do, but anything more needs an interface to the firmware that doesn't exist yet, much bigger job.

PR #3163 (2015-12-18): Projectkk2glider/issue 3160 simulator commandline
…or (radio type and eeprom file)

Help build in :wink: 
```
$ ./simulator21 --help
Usage: simulator [OPTION]... [EEPROM.BIN FILE] 
Options:
 --radio[=]x, -r[=]x  radio to simulate 
 --help, -h           show this help text 

Available radios:
	opentx-9x
	opentx-9x128
	opentx-9xr
	opentx-9xr128
	opentx-9xrpro
	opentx-ar9x
	opentx-gruvin9x
	opentx-mega2560
	opentx-sky9x
	opentx-taranis
	opentx-taranisplus
	opentx-taranisx9e
```

Closes #3160
commit 0bfcd845 (2015-12-18): Fixes #3160: command line options support added to standalone simulator (radio type and eeprom file)
commit 0db5072b (2015-12-18): Fixes #3160: command line options support added to standalone simulator (radio type and eeprom file) (ported from master)
closed 2015-12-18 by bsongis