opentx/opentx #3161: X9E/Companion 2.1.5 Connect Problem

qqwasder, 2015-12-17:
I cannot connect from linux to my X9E, for example:
  Two lower trim switches in, hit power on - Taranis bootloader message, etc
  Connect X9E to linux PC via usb
  File manager pops up and directories are presented (including their contents), exit file manager
  Start Companion 2.1.5 (the results are the same for 2.1.3)
  Read/Write > Read Models and Settings From Radio
  "No Radio was found!..." - Hmmmm
Run dfu-util -l from command line - nothing presented
Run lsusb: "...Bus 001 Device 006: ID 0483:5720 STMicroelectronics..." displayed
Repeat above using Windows 7 - all aspects of everything work perfectly
I am at a loss for a direction to proceed. Any help you can provide would be greatly appreciated.
   Thank you...

projectkk2glider, 2015-12-17:
Do you see both drives? One is SD card, other (this one is used by Companion) contains eeprom.bin and firmware.bin files.

qqwasder, 2015-12-17:
Yes, I see both when the file manager pops up. One is listed as TARANIS 
directory containing 2 files: EEPROM.BIN and FIRMWARE.BIN; the other is 
listed as 952.4 MiB Removable Media and contains 8 directories: BMP (587 
items), EEPROMS, FIRMWARES, LOGS, MODELS, SCRIPTS (6 items), SOUNDS (6 
items), and System Volume Information.

projectkk2glider, 2015-12-18:
The Companion searches on all mounted filesystems for the EEPROM.BIN file. Check your access permissions and where the Taranis virtual drive (the one containing EEPROM.BIN) is mounted (use `mount` command in terminal). Perhaps the drives are not mounted conventialy on your system and file manager is showing them via some mechanism (like fuse, smb, etc..).

qqwasder, 2015-12-18:
You nailed it! I'm using openSuSE 13.2. When connected, the X9E shows up 
as /dev/sdj mounted to /run/media/username/TARANIS. After umount 
/dev/sdj, mount /dev/sdj /mnt, companion21 runs just fine. Thanks so 
much for your insight...

closed 2015-12-19 by projectkk2glider