opentx/opentx #3162: getFlightMode() troubles

jfrickmann, 2015-12-18:
getFlightMode() is not working. In the little test script below, I get a nil value error, in my bigger script I get a syntax error. But if I add a local getFlightMode() function, then it works. I am testing on Companion 2.1.6. I hope that someone can help - thanks!!

Example:
-----------------
local x

local function init()
end

local function background()
  x = getFlightMode()
end

local function run(event)
  background()
  lcd.clear()

  lcd.drawText(16, 16, x)
end

return { init=init, background=background, run=run }
-----------

projectkk2glider, 2015-12-18:
This function will be in 2.1.7, see #3127 and https://opentx.gitbooks.io/opentx-lua-reference-guide/content/general/getFlightMode.html

closed 2015-12-18 by projectkk2glider

jfrickmann, 2015-12-18:
I did not know that I was that cutting edge :-)

Thanks!!!

---------------------
Jesper Frickmann