opentx/opentx #3167: Add the support for jpg and/or png

projectkk2glider, 2015-12-20:
Parent issue #3159

projectkk2glider, 2015-12-20:
Classic PNG implementation:
* reference implementation in libpng http://www.libpng.org/pub/png/libpng.html
* also requires zlib http://www.zlib.net/

All in one alternative:
* lodepng https://github.com/lvandeve/lodepng

----
Jpeg:
* STM32CubeF4 from ST has Jpeg library http://www.st.com/web/en/catalog/tools/PF259243

projectkk2glider, 2015-12-26:
A couple of questions for @bsongis:
* do we need read only capability or write (create) also?
* what kind of format should be returned from the read function - ie RGB, how many bits per color, etc..
* maximum image dimensions

bsongis, 2015-12-26:
I suggest we add a byte to remind the format of the loaded bitmap. It will save CPU

bsongis, 2015-12-27:
- Write support would be a bonus and only on PNG, if implemented, I think it would be good to have snapshots in PNG format rather than BMP
- Ideally the format returned would be the one which would be the quickest to display, (without loosing quality of course).  I suggest we add a byte to remind the format of the loaded bitmap, just near the width and height. It will save CPU
- Maximum image dimensions. Given we have plenty of RAM, it would be the LCD size, no? The malloc should use the external RAM of course.

projectkk2glider, 2015-12-30:
I will look into PNG tonight....

projectkk2glider, 2015-12-30:
My current favourite is https://github.com/nothings/stb/blob/master/stb_image.h which supports loading a bunch of formats, we can chose which ones we want, others are not compiled in.

bsongis, 2015-12-30:
Seems good!

projectkk2glider, 2016-01-03:
More resources:
* https://news.ycombinator.com/item?id=8600634
* jpeg compressor https://github.com/richgel999/jpeg-compressor

projectkk2glider, 2016-02-28:
This can be closed now, done.

closed 2016-02-28 by projectkk2glider

RCdiy, 2016-09-08:
So now we can use a rgb PNG file with transparency without needing to index it to 16 colours?

kilrah, 2016-09-08:
Yes