opentx/opentx #3168: USB Joystick with 9XR Pro

AlexDaniel, 2015-12-21:
[It seems like](https://github.com/opentx/opentx/wiki/OpenTX-USB-Joystick) this feature is supported on Taranis but not on 9XR Pro.

LeoDJ, 2016-06-12:
push?

schwabe, 2016-06-12:
@leodj someone need to implement that feature. And as in every OpenSource project most people do what they like to do.

buzztiaan, 2016-11-19:
+1 

anyone have a idea whats needed for this?

kilrah, 2016-11-19:
Someone needs to write the USB driver and application code for the 9XR-Pro's processor (or find an example to modify).

dandruff39, 2016-11-19:
Are these reasonable references for what's required (supplemental to the Taranis joystick code)? 
http://www.atmel.com/products/microcontrollers/ARM/SAM3S.aspx?tab=documents
http://www.atmel.com/Images/Atmel-42339-USB-Device-Interface-UDI-for-Human-Interface-Device-Generic_ApplicationNote_AT09334.pdf

mayanez, 2017-06-12:
Any update on this?

schwabe, 2017-06-12:
@dandruff39 yes. @mayanez no.

AlexDaniel, 2019-08-17:
It is not stale, there's actually a PR fixing this: https://github.com/opentx/opentx/pull/5776

PR #5776 (2019-08-26): USB joystick support for 9xrpro
Fixes #3168

This code is written based on the current joystick implementation for the Taranis and the example for HID mouse from [here.](https://github.com/lfwanner/var/tree/e9802710380a480fa91a6660e754f3a087be5484/sam3/at91lib/usb/device/hid-mouse)

Tested on Turnigy 9XR PRO. As I understand, it will work for all transmitters with SKY9X mainboard.

Also, masstorage for SKY9X removed, because it is no longer used (as i understand, it is used now only by the bootloader, which is downloaded from an [external repository](https://github.com/MikeBland/OpenRcBootload [... cut]
closed 2019-08-26 by bsongis