opentx/opentx #3170: Feature Request: Ability to upload FAA digital cert image into radio

eep357, 2015-12-22:
With the passing of mandatory FAA registration in the US, digital or printed copy of the certificate is required to be on hand when flying. Storing this in the radio would ensure compliance without needing phone with internet available while in the field

kilrah, 2015-12-22:
Would be unreadable on the radio's screen. But you can put the PDF on your phone, no need for internet connectivity then.

projectkk2glider, 2015-12-30:
You can store it on the radio (on the SD card) right now, you just won't be able to display it with the radio (unless it is in text or BMP format). If you can store the certificate in a plain text file, then it can already be displayed by the radio.

PDF support on the radio is out of the question. There will be support for colour bitmaps (jpg on png format) on Hours, but again the size will be limited to the display size. I really don't see how this can be elegantly solved by OpenTX. I suggest you follow Kilrah's recommendation.

closed 2015-12-30 by projectkk2glider