opentx/opentx #3171: Nightly builds. Playback of log file triggers logging.

dwheeld, 2015-12-24:
It is possible for an event in the log file being played back to trigger logging. This adds the playback period onto the end of the log file being played. Using altitude above 30 feet to trigger logging is one case noticed.

kilrah, 2015-12-24:
Don't understand your post. Is it a request? An observation? A bug report?

dwheeld, 2015-12-24:
No, yes,yes.

kilrah, 2015-12-24:
OK, I think I got it, but please try to be a bit descriptive next time...

I translate that into:

> I'm using the telemetry log file player in the companion simulator with a model that has logging configured to trigger when Alt >30ft selected, and as the log file is playing and sends altitude values >30ft the simulator starts logging and appending a copy of the played data at the end of the file.

Now we can understand the issue and it makes sense. 

Not sure what the correct course of action is there, disabling the logging SF when the player is playing requires an interface between simu and FW that are currently independent.

dwheeld, 2015-12-24:
It's a minor bug, to be sure. However I just wanted to let you know so that if possible, it could be squashed before the next release. 

A simple work around is to make a copy of the model and delete the logical switch that triggers logging. This new model can then be used to play back log files.

projectkk2glider, 2015-12-30:
Actually I see no problem. Why do you play-back the same file that is currently used as an logging output? Just make a copy of the log file and play it (the copy). No data should be appended then.

closed 2015-12-30 by projectkk2glider

kilrah, 2015-12-30:
Well the fact that it creates a new useless log could be considered as one - but indeed very minor and not worth bothering IMO.