opentx/opentx #3172: logical switch for channel seems to run after SF overrideCH and returns incorrect result

marcmerlin, 2015-12-24:
Tested in companion 2.0.15 although the issue seems to be in opentx itself.
From what I can tell, it's not a bug per se, but a dependency problem that OpenTX might be able to be improved to work around?

Repro:
1) set S3 as 100% mix for channel 12
2) SF13: SFdown override CH12 to 100
3) LS19 fun::a>x, V1:CH12, V2: 1

S3 is 0 and I can't move it in companion anyway since the S3 pot isn't emulated. CH12 is 0 too
I flip SF to down, CH12 goes to 100%
LS19 does not toggle

My guess is that LS19 is computed before the SF override happens.
I realize in most cases you need to compute LS before SF runs, although in my case I needed the opposite. 

In my case, CH12 is my flight mode channel. It's linked to my S3 pot, but there are conditions that cause an override of CH12 to -100 (RTL) or 100 (MANUAL).
If CH12 is anything other than 100, I need to disable some mixes that mess with ailerons and elevator when I'm in manual control, so I was going to make them conditional on LS19.
But as discussed, there may not be a way to have a LS read a channel, or even a GVAR if that value may be modified by a SF if OpenTX has a fixed ordering between LS and SF.
Is there any way out of this chicken and egg scenario, or is it just not something the software can support?
If not, feel free to close this issue.

kilrah, 2015-12-24:
Override is indeed a clamp that is independent of everything else.
The standard way to do what you want is to use a Replace mix as last line of your CH12.

closed 2015-12-24 by kilrah

marcmerlin, 2015-12-25:
On Thu, Dec 24, 2015 at 10:52:13AM -0800, Andre Bernet wrote:
> Override is indeed a clamp that is independent of everything else.
> The standard way to do what you want is to use a Replace mix as last line of your CH12. 

That makes sense, I appreciate the answer.
Is there documentation I can update myself to add this hint/warning to
people using overrideCH, or are the docs handled by someone with special
access?

Thanks,
Marc
-- 
"A mouse is a device used to point at the xterm you want to type in" - A.S.R.
Microsoft is to operating systems ....
                                      .... what McDonalds is to gourmet cooking
Home page: http://marc.merlins.org/                         | PGP 1024R/763BE901

kilrah, 2015-12-25:
Documentation is a bit of a mess at the moment so there isn't really a good place to just add such a thing to...

marcmerlin, 2015-12-25:
On Fri, Dec 25, 2015 at 08:28:56AM -0800, Andre Bernet wrote:
> Documentation is a bit of a mess at the moment so there isn't really a good place to just add such a thing to...

Gotcha. If all goes well Google should index this anyway, so the next
person has a better chance to find the answer.

Thanks again,
Marc

bsongis, 2015-12-25:
The doc is also on github, if you want to give a hand you may open a fork, make your mods and open a pull request

bsongis, 2015-12-25:
Taranis manual:
https://github.com/opentx/opentx-taranis-manual

bsongis, 2015-12-25:
And the result is there:
https://opentx.gitbooks.io/opentx-taranis-manual/content/

marcmerlin, 2015-12-26:
Merci Bertrand @bsongis 
I'm thinking about writing the following
"Override CH in SF happens at the end of all processing, which means that if you have a logical switch based on a channel value that you override in a special function, you will get the wrong channel value.
More generally, it's more flexible to use a replace mix in your channel definition than override CH in a SF"
If that sounds good, I'll find a good place to add this to the docs and send a pull request.

bsongis, 2015-12-26:
It sounds good to me

marcmerlin, 2015-12-29:
@bsongis Bertrand, I had a look at the manual you gave me, but it doesn't even mention anything about special functions, so it's hard to add a warning about a specific use of one CF/SF when CF/SF aren't even mentioned that all that I could tell :)
(mmmh, I now see that special functions used to be called custom functions)

For what it's worth, I learned opentx with
https://docs.google.com/document/d/1qlh09LzxtpPt7j_aqG8yiOu2yoYMzP9XA-PJA81rDJQ/edit

Do you think it could be merged into the wiki? It's pretty complete and better than rewriting all that stuff from scratch.
Either way, we totally left the purpose of this issue now, so feel free to ignore further replies here, but I just wanted to get back to you about why I couldn't update the github wiki.

Best, Marc

bsongis, 2015-12-30:
Custom Functions have been renamed to Special Functions since long now!
It would be nice to copy / paste the best from these 2 documents into the new GitBook document. Would you help us for this? It would be great, we are not so many and we have too little time to do everything right on due time ...

lshems, 2015-12-30:
I want to do my bit there, as I already mentioned. Just pm me if you want
to get into it as well, so we can discuss how to approach this.

kilrah, 2015-12-30:
Thats what I meant when I said it was a bit of a mess.
There is the (pretty complete, but outdated) r2940 manual linked above. Then there is a partial update of that to 2.1 that was never finished. Then the Gitbook is a partial copy of that, that was never finished either. 
So right now working on the gitbook repo is basically working of an incomplete copy of an incomplete refresh job of an outdated doc.

IMO ideally the gitbook should be first brought to the state of a comprehensive 2.0 manual, then make a copy of it and add the new 2.1 stuff.

marcmerlin, 2015-12-30:
Bertrand, while I had an hour or so I could volunteer, this is a *much* bigger project than that, more in the multiple days of work range (I know, I've already done a few of those :) ), and I'm already busy with a few other projects.
Actually from what I remember, just writing
https://docs.google.com/document/d/1bGKjyvOksHBhD8gT3Q20ZmDu1TUsDn15BvGPHw-doVg/edit
took me about 2 days of work :)
If I end up with spare time on my hands a bit later, I could reconsider this, but right now it wouldn't be reasonable for me to volunteer on one more thing :)

marcmerlin, 2015-12-30:
I should add that's it's totally cool for anyone to copy part of all of my doc to use in any opentx documentation though.