opentx/opentx #3173: Global Variable

frobert0, 2015-12-24:
Spezial Funktion
SF2   Adjust GV1  Wert   85
Logische Schalter
L 1  a>b  [i3Gas]   GV1

>> bei ~ 8% Throttle aktive L1

---------------------------------------------------------------

Logische Schalter
L1 a>x  [i3Gas]  85

>> bei ~85% Throttle aktive L1

?????????????????????????????????????????????

kilrah, 2015-12-25:
https://github.com/opentx/opentx/issues/1361

closed 2015-12-25 by kilrah

frobert0, 2015-12-26:
für Programmierer und Mathematiker, nicht für Normale, es solte immer der selbe wert sein nicht einmal so und einmal anders, wenn Funktion benötigtig dann sollte sie anders benannt sein

kilrah, 2015-12-26:
Again #1361 explains why it is that way and that it will eventually be changed for what you want.