opentx/opentx #3178: Interrupt priority potential problem

MikeBland, 2015-12-27:
Taranis/Horus:
File usb_bsp.c
Function: void USB_OTG_BSP_EnableInterrupt(USB_OTG_CORE_HANDLE *pdev)
This function calls:
  NVIC_PriorityGroupConfig(NVIC_PriorityGroup_1);
Depending on how you expect the interrupt priorities to work, this has the potential to completely upset them. The default operation is to have 16 "group" priorities thus an interrupt of priority 0 will interrupt anything else. The above call changes this to having only TWO groups (0 and 1), with 8 sub-priorities within each. This means interrupt priorities 0 to 7 are one group and 8-15 are another. An interrupt of priority 0 will no longer interrupt those of priority 1 through 7, only those of 8 through 15. Also priority 1 won't interrupt 2 through 6 etc.

Function: void USB_OTG_BSP_DisbleInterrupt(USB_OTG_CORE_HANDLE *pdev)
also calls this.
Unless you want the interrupt priority grouping changed, both functions need this call removing.

I would suggest:
void USB_OTG_BSP_EnableInterrupt(USB_OTG_CORE_HANDLE *pdev)
{
	NVIC_SetPriority( OTG_FS_IRQn, 11 ) ; // Lower priority interrupt
	NVIC_EnableIRQ(OTG_FS_IRQn) ;
}

void USB_OTG_BSP_DisableInterrupt(USB_OTG_CORE_HANDLE *pdev)
{
	NVIC_DisableIRQ(OTG_FS_IRQn) ;
}

Mike.

kilrah, 2015-12-29:
How did you notice the problem, i.e. what undesirable operation did it cause? To get an idea if it any existing issues could be cuased by it...

We've had the joystick code in there for about a year and a half now.

MikeBland, 2015-12-29:
I found this using ersky9x. I have interrupt driven code that allows the SPort and the serial port to receive non-inverted data. This relies on a high priority interrupt (level 0) to capture the transisitions of the serial signal. I recently added the joystick code in to ersky9x (STM platform). When testing the software serial for use with telemetry from the MULTI module (DIY), it was missing transisitions, which it didn't use to do on telemetry from the Orange DSM telemetry module. (The interrupt is generated by an edge on a GPIO signal, the interrupt then reads the value of a timer/counter as no timer/counter hardware capture is available on the specific pin).
The joystick code meant all interrupts given priorities 0 to 7 become the same pre-emptive priority, so interrupt level 0 no longer interrupted levels 1 through 7.
Example, the 5mS interrupt is an interrupt that can take a significant time to execute (openTx level 7, ersky9x level 4). If this interrupt is executing, then all other interrupts of level 0 to 7 will not execute until this one completes.
Any other undesirable effect depends on the time taken to execute interrupt routines, and if this time could cause another interrupt to be delayed too long. e.g. a serial port receiving data at 115200 baud (and I've also seen a referrence to "crossfire" at 200k/400k baud), that doesn't execute a receive interrupt within 85uS could suffer an overrun error (25uS at 400kbaud).

Mike.

projectkk2glider, 2016-04-10:
I have checked this. Mike is right. The reset value of interrupt priority groups is one group with 16 levels. The old USB code changes that - we do not want that. The fix was already shortly tested on Taranis (master) and Horus (next).

commit 14841590 (2016-04-10): Fixes #3178: USB driver changed interrupt grouping which affected interrupt priorities (fix suggested by Michael Blandford)
PR #3410 (2016-04-29): Fixes #3178: USB driver changed interrupt grouping which affected int…
…errupt priorities (fix suggested by Michael Blandford)

I recommend not merging right now for 2.1.8. It was tested on my Taranis, but should be tested some more time on the next branch (changes there done in commit 14841590b9e092d6daf7056426c449e601c15b12).

Closes #3178
commit 4a05c2f6 (2016-04-29): Merge pull request #3410 from opentx/projectkk2glider/issue_3178_for_master

Fixes #3178: USB driver changed interrupt grouping which affected int…
commit 5b2694eb (2016-04-29): Fixes #3178: USB driver changed interrupt grouping which affected interrupt priorities (fix suggested by Michael Blandford)
closed 2016-04-29 by projectkk2glider