opentx/opentx #3179: TRx  dont work if not pluged a 3.5mm in the trainer ( taranis)

ArielBaravalle, 2015-12-28:
Hello.
I designed this  [stickPlus] ( http://www.picopters.com/es/27-stickplus.html)  ,  the stickplus send a ppm signal of 8ch  to the trainer input of any RC emitter,   and mix  insert the desired channels in the output to control gimbal or pantilt cameras.
the connection is to the inside of the taranis whit the [image:](http://tapatalk.imageshack.com/v2/15/02/10/ba1b553618a2fa21820a23eb2056e06a.jpg)  and in the pix  can use the TRx to insert this channels into a desired  output.
after the las change of version in Opentx from 2.1.3  to 2.1.5  , this connection don't work.

after try diferent options I discovered that work if i put a 3.5mm connector in the trainer,  and enable the internal TX in any mode.

why the system not read the ppm trainer if don't have a plugged trainer?  can enable a option to permit this read?

thanks.
Regards
Ariel Baravalle

kilrah, 2015-12-28:
This was done to solve #2722.
When nothing was connected to the trainer input (left floating) then external RF noise could be picked up and considered as a valid signal. So the plug deteciton is now used as it should to enable/disable trainer input altogether.

closed 2015-12-28 by kilrah

projectkk2glider, 2015-12-28:
Connect this pin to groud to enable PPM in trainter function:
[image]

ArielBaravalle, 2015-12-28:
Ok. But...
Is not  more simple a on-off option in the trainer configuration screen ?

ArielBaravalle, 2015-12-28:
Or if no use any mix with TRx or trainer,  automatic disble this ppm input.
And all happy .  :)

Airmongrel, 2015-12-28:
Hi,
I have this problem with my stickplus since updating from 2.1.3 to 2.1.6. I've read the above posts, but is there not a way to be able to turn on or off so this sill resolves issue #2722.

Thanks in advance,

Mongrel.

projectkk2glider, 2015-12-28:
Why add a new settings (which would require an EEPROM change and thus could only make it into 2.2) when you can simply ground the pin I outlined above. Alternatively you could just plug in the empty stereo jack and achieve the same effect.

kilrah, 2015-12-28:
Yep... if you're soldering the device inside (=permanently attaching it, preventing use of the trainer port anyway) I don't see the problem with tying the detection line as well.

ArielBaravalle, 2015-12-28:
Ok,  no problem.
Yes ,  is a good solution.  Thanks and great work.

ArielBaravalle, 2016-02-19:
I tryed to put to ground the pin indicated in the previus image,  and the trainer input don't work.
With this bridge the trainer icon show constantly in the display.
[image]