opentx/opentx #3180: Feature/Bug:  Allow servo travel speed to use a GVAR in 2.1.x

mcddd, 2015-12-28:
Currently servo travel speed can be adjusted from 0 to 20 seconds, but does not allow the option to select a GVAR.  The ability to use a GVAR for servo travel speed and delay times would allow functions such as flaps and retracts to operate at different speeds based on a switch position, without the timing issues described in the following post from the 2.1.x thread.

http://www.rcgroups.com/forums/showpost.php?p=33522564&postcount=2234

kilrah, 2015-12-28:
From my understanding you'd want to be able to switch speed during the operation, which this would not allow for. 
Why not have your "fast" speed just not have a slow? Sounds like you'd only want it for emergencies anyway.

mcddd, 2015-12-28:
I want the mechanical retracts to be able to operate at two different speeds.  The fast speed is to allow the gear to be quickly retracted or extended.  The slow speed matches the time it takes the gear to retract or extend on the full scale plane.  So the gear can retract or extend at a slow scale-like speed, or the gear can be quickly retracted or extended.  The same situation could apply to operating flaps.

So if a GVAR isn't the answer, then what is the solution?

kilrah, 2015-12-29:
Well you've got 2 solutions in that thread, mine that gives you a "precise" timing using the slows but with the drawback that changing speed while a movement is being processed will cause a jump, and Mike's solution that doesn't have that problem but requires you to play with mix percentages to find the speed that pleases you.

closed 2015-12-29 by kilrah