opentx/opentx #3181: As designed or BUG? Using special function to adjust gVar with trim buttons limits to 125

lshems, 2015-12-29:
when using a special function to change a global variable and setting a channel trim as input, I would assume that the range would match the -1024 - 1024 range of a gVar.

Instead, it is limited to the -125 - 125 range of the corresponding trim.

Since the trim value itself is invisible and not adjusted when having such a special function activated, I see no reason why it should not simply increment and decrement the gVar value on trim button push, and limit it to -1024 - 1024??

kilrah, 2015-12-29:
Because imagine how long it would take to "scroll" the whole range.
You can use the trim *keys* instead of the trim *register* to trigger an "Adjust GVx +1" (and -1) SF if you want no boundary.

closed 2015-12-29 by kilrah

lshems, 2015-12-29:
Not if I want it to be active only an LS condition.

The how long is not an argument in my opinion, since if you don't like the 'too long', use your solution, with the restrictions.

Use case: when flipping a switch (or LS), allow trim buttons to easily adapt gVars used as mix percentages. Because no stick inputs are used, this enables in flight adjustment of ALL mixes, if properly done, by only using switches and trim buttons. No need to enter the menus, and a very nice shortcut.

The advantage is equally useful for all other uses of gVars, where in flight adjustments need to be easily accessable. 

In my case I use it for flight-mode dependant camber settings on F3K gliders. There is almost no easier way than this to do in flight tuning, and it is very intuitive, using the trims on the flaperon control to adjust camber.

P.S. I can work around the 125 limit, but I still see no reason why it would/should be limited. It is a reassigment of the function of the trim buttons, it is not an actual use of the trim register, as that  remains unaltered when the SF is active. 

I can understand that it might not be an obvious code change, as I think you now only map the gVar register to the function, instead otf the trim register, and my request would need some modifictions beyond that.

kilrah, 2015-12-29:
> Use case: when flipping a switch (or LS), allow trim buttons to easily adapt gVars used as mix percentages. Because no stick inputs are used, this enables in flight adjustment of ALL mixes, if [...]

Which works perfectly right now, I've used it myself - everything to do with mixing and offsets works on the +/-100 range unless it's a very special case. That's precisely why it was done that way...

lshems, 2015-12-29:
Yes, if you don't need more resolution in your mix settings as about 10% of
set point value.

It's not logical to limit the inputs on gvars to 125, just because they are
changed with the trim buttons.

But I understand there are other points of view, which I respect.

Thanks for your efforts, and your patient feedback is really appreciated!

bsongis, 2015-12-29:
This is a problem which is difficult to solve.
Two proposals for OpenTX 2.2:
1) what about adding an unit to a GVar? It would solve our problem with GVars used for channels. Some like to use 512us, other 100%, other 1024(raw)
2) what about adding a AdjustGVar (+/-X) instead of AdjustGvar(+/-1)
The purpose is to unify everything...

kilrah, 2015-12-29:
> Yes, if you don't need more resolution in your mix settings as about 10% of set point value.

Why? Percentages are mapped 1:1 when used in a weight/offset etc. You get the entire range. 

>1) what about adding an unit to a GVar? It would solve our problem with GVars used for channels. Some like to use 512us, other 100%, other 1024(raw)

And prec then, to take care of #1361 / #905... Gonna need a major UI redesign.

>2) what about adding a AdjustGVar (+/-X) instead of AdjustGvar(+/-1)
The purpose is to unify everything...

Yes the AdjustGVAR +/-X is way overdue (#1713)

lshems, 2015-12-29:
>>
>> Yes, if you don't need more resolution in your mix settings as about 10%
of set point value.
>
> Why? Percentages are mapped 1:1 when used in a weight/offset etc. You get
the entire range.
That's exactly the problem. Say your mix is useful in the range of 20- 50 %
then you have a resolution of only 30 steps, so 3% In practice, I need
mixes in closer ranges, so I end up with only 10% steps, which is way to
coarse. That's why I multiply by 10% in my mixes when using gvars.

Sounds like the unit problem there!
>>
>> 1) what about adding an unit to a GVar? It would solve our problem with
GVars used for channels. Some like to use 512us, other 100%, other 1024(raw)
>
> And prec then, to take care of #1361 / #905... Gonna need a major UI
redesign.

bsongis, 2015-12-30:
Yes, PREC as well. I will open a new issue for those ones. These 3 points need a EEPROM change, that's why it has to wait for 2.2

bsongis, 2015-12-30:
Refactoring explained in #3185