opentx/opentx #3182: Taranis Firmware 2.1.6 - Timer not reading value correct!

wot4, 2015-12-29:
I recently updated my Taranis+ from 2.0.13 to 2.1.6.

I have noticed that all of the Telemetry 'logical switches' and 'Special Functions' have stopped working.  

For example I had (2.0.13):
a<x   Cells   10.5v

Now the transmitter displays it as (2.1.6):
a<x   ---   105

Also I had a switch to read the remaining time on the timer1, now this doesn't get read properly.

For it to read the time 12.50, it says "12.0 and 50.2". Its also doesn't include the "mins" word at the end of the speech.

Any suggestions ?

kilrah, 2015-12-30:
>I recently updated my FrSKY Taranis+ from 2.0.13 to 2.1.6.

...aaand like many you forgot to update the sound pack along with it.

closed 2015-12-30 by kilrah

projectkk2glider, 2015-12-30:
Read http://www.rcgroups.com/forums/showthread.php?t=2451551

wot4, 2015-12-30:
Thanks for the feedback.

Sound files working correctly now....