opentx/opentx #3183: No Haptic - 2.1.6 - Taranis X9D Plus

dWoolridge, 2015-12-29:
I just upgraded to 2.1.6 and just noticed that my transmitter no longer produces haptic vibrations.  It does make a very faint electronic sound when I would expect the haptic to be active.  Please note this is not a mechanical sound I am hearing.

I'm somewhat new to Open TX, so hopefully I'm not missing something simple.  I created a special function on a switch to test the Haptic.  The transmitter produces the sound described above when the switch is active, and the sound pulses 3 times as I set the Haptic pulse in the special function.

General Settings:
----------------------
Haptic mode:  ALL  (also tried No Keys)
Haptic length:  normal
Haptic Strength:  I've tried all Haptic Strengths

Thanks,

Doug

projectkk2glider, 2015-12-30:
Can anybody verify this. I suspect it is a user setting problem somewhere.

@dWoolridge you upgraded from which version? Or to put it differently, which was the last version that worked for you?

dWoolridge, 2015-12-30:
I know it worked with the firmware that was on the transmitter when I bought it new.  I can't say if it ever worked since updating the firmware.  I just discovered how the haptic would be nice to use, and realized that it wasn't working.  The firmware currently reported by the transmitter is "opentx-taranis-plus" version 2.1.6 (2015-11-12 01:10:06)  EEPR 217.

In addition to the "General settings" I noted above, I've been using Special Function 8 (shown below) to test.  If I hold the transmitter right to my ear, I can hear 3, faint, electrical "pulses" when I flip SA to the up position.

[image]

dWoolridge, 2015-12-30:
One other thing I noticed... but I'm not sure it would help.

I don't see "Haptic" as a build option when I go to Edit Settings.  Based on something I read somewhere, it sounded like that may have been an option in previous versions.  (see below)

[image]

kilrah, 2015-12-30:
Normal because Taranis plus always has haptic, so no option needed. The menu options wouldn't appear if it wasn't enabled.

projectkk2glider, 2015-12-30:
Any chance you have your Haptic strength setting on minimum?

dWoolridge, 2015-12-30:
I've tried all 5 haptic strength settings.

dsbeach, 2015-12-30:
I can test tomorrow.

dsbeach, 2015-12-31:
Tested today on 2.0.15 and then on released 2.1.6 and an up-to-date self built master.  No issues found.  At lowest strength the haptic barely noticeable over the sound from the speaker (coming from my 'play track'). I did not use a LS but otherwise I think the test is valid.

dWoolridge, 2015-12-31:
Thanks for testing that.  2 quick questions...

1)  Where can I get just the .BIN files  for previous versions of the firmware?  I'd prefer to try a previous version by doing that rather than installing another version of Companion to get the firmware.

2) Does the companion simulator simulate the haptic vibration in any way?

dsbeach, 2015-12-31:
Sorry - I don't know if the older .bin files are available.  Also, it appears that there is no haptic feedback in Companion.

powercycle, 2016-01-02:
Haptic works for me on 2.1.6, eepr v 217.  Are you positive that you are using the Taranis Plus profile?

If not, it could cause issues similar to this.

dWoolridge, 2016-01-02:
I believe so...  under settings it says "Radio Type->FrSky Taranis Plus".  Is there anywhere else I need to confirm this?

Also, I just popped the radio apart and applied 1.5V directly to the haptic motor solder points.  No result.  I then tried 4.8V and still nothing.  Do you know if that voltage should have spun the haptic motor?

powercycle, 2016-01-02:
Guess it could just be a hardware issue.

kilrah, 2016-01-02:
Sounds like your haptic motor is simply dead.

closed 2016-01-02 by kilrah