opentx/opentx #3185: GVARS redesign

bsongis, 2015-12-30:
As stated by many issues. GVars would benefit of 3 more parameters:

[] Add an unit. It would solve our problem with GVars used for channels. Some like to use 512us, other 100%, other 1024(raw)
[] Add a prec. It would take care of #1361 and #905
[] AdjustGVar (+/-X) instead of AdjustGvar(+/-1), people wait for this function since long, cf #1713

bsongis, 2015-12-30:
GVars are needed in some more menus:
[ ] Audio Volume #2248
[ ] Slow / Delay

hrenz, 2015-12-30:
Hy Bertrand,
would be very nice!
Thanks!!

arronbates, 2015-12-30:
very cool

projectkk2glider, 2015-12-30:
There is also one existing subtle bug related to gvars when using FUNC_ADJUST_GVAR: 
* the increment/decrement is missing a limit check. The value of gvar can thus be incremented or decremented beyond the normal range.

kilrah, 2015-12-30:
That was also an issue I think, someone wanted to be able to set a limit to adjustment range. Will see if I can find it.

hrenz, 2015-12-30:
Hy,
I think about GVAR 
Min, Max, Start value, Ind/Dec Steps, Typ % or us

fraternl, 2016-01-03:
@bsongis 
I haven't been following developments in the last 2 months and noticed these changes in gvar.
From what I gather I can soon change a gvar using a stick as a direct source and let it change a gvar within my desired range.
This means that one doesn't have to use a helper channel with a dedicated curve to achieve this (unless I want even more control).
Am I correct in this?

http://www.rcgroups.com/forums/showpost.php?p=33597465&postcount=2349

kilrah, 2016-01-03:
No, the idea would have been to use the new limits as a clamp, not scaled to the input.

Reason is that if we scaled and that's not what the user wants it's hard to "undo" the scaling, while the current way of using a channel makes the opposite easy.

kilrah, 2016-01-03:
Another GVAR issue: #3103

bsongis, 2016-01-03:
https://files.gitter.im/opentx/opentx/MRgv/Sn_mek-obrazovky-po__zen_-2016-01-03-21_03_43.png

bsongis, 2016-01-13:
Some progress on this in my last commit. GVars names are now reduced to 3 chars. Curves as well. Names for GVars and Curves will be used everywhere in the firmware now.

Tests needed please!

lshems, 2016-02-01:
any suggestions for tests?
Able to do some.

hmichl, 2016-02-12:
I really like this thread and support the idea of having GVARs at additional places. One request I would like to add is having also the possibility to have a GVAR instead of a value at the weight of trainer-functions. I want to be able to change this weight dependend on the position of the stick of the master-transmitter (for example the more I move my stick to a maximum position the less weight the student has). I already implmemented this functionality with a lot of mixers using TR1 to TR4, but having it where it belongs to (in my opinion :-) would be great!

In addition to this, it would generally be cool to have the choice of either GVARs or the value of a channel. I often use channels and functions to generate values, but then have to assign exactly these values to a GVAR. Would be cool to skip this one step.

RCdiy, 2018-02-06:
I've defined LS1-5 like this. Ignore L1.
[image]
GV1 changes with stick THR
GV2 is 50%
GV3 is 50
When comparing THR against GV2 and GV3 should the behaviour not be different?
When THR is equal to 5 why is it equal to both 50% and 50?
I thought THR had unit % so 5 should not be equal to 50%.
[image]
[image]

Here is the OTX file. Rename and remove .txt to use it.
[GVARtests.otx.txt](https://github.com/opentx/opentx/files/1700805/GVARtests.otx.txt)

kilrah, 2018-02-07:
Ah I believe the % is only for display, you need to set the "precision" for actual scaling.

closed 2020-04-06 by stale[bot]