opentx/opentx #3187: Error reading switch label from X9E

jvmatern, 2015-12-30:
When downloading radio settings to companion is switchK "SK-" given the wrong name "!FM8".
Companion 2.1.6
open-tx-taranis-x9e 2.1.5

[image]
[image]

mpaperno, 2017-04-11:
This is actually a radio firmware bug.  When a switch in a 3-way "slot" is configured as two-way, the wrong value is registered when auto-detecting switch position on the radio.  

Two-way switches should register as UP/DOWN (arrows).  That is how it acts in eg. slot SF.  But if you configure any 3-way slot as 2-way, then the auto-detection shows only `-`/DOWN (no UP arrow).

If you manually scroll through the available switch positions on the radio, then only the UP/DOWN options are shown, there is no `-` middle position option at all.  Which seems correct.  So only auto-detection seems broken in this case.

This is not limited to the LS screen, but seems to be throughout the X9E radio UI.   Nor can I recreate this on any other radio/UI.

3djc, 2017-04-11:
I have a bug registered for this, but this won't get looked at before 2.2.0 I think

mpaperno, 2017-04-11:
OK, perhaps close this one then if it's a dupe.

3djc, 2017-04-11:
Yeap and the real issue is with the radio

closed 2017-04-11 by 3djc