opentx/opentx #3189: More type of switches in harware settings

fraternl, 2015-12-31:
I have replaced 2 of my switches by different types that I can't choose in hardware settings.

I have replaced a standard 3-position switch with a 3-position switch that's only on 1 side latching. The other side is momentary. They are sometimes referred to as (ON)-OFF-ON
The other switch I replaced is of type (ON)-OFF-(ON) which means it will always goes into neatral (OFF).

AFAIK the simulator follows these settings (I can't really test as Companion started to become extremely slow in the latest nightlies) and it would be nice if it would follow that changed behaviour.

kilrah, 2015-12-31:
Not really a fan of adding settings to the firmware that do absolutely nothing in it...
And no, the companion simu corrently doesn't take any of the switch settings into account.

MicheleVilla, 2015-12-31:
There is another open issue about that
#2960

hrenz, 2015-12-31:
Hy.

we will need a   2 side Toggel Switch  (ON)-OFF-(ON) option with autorepeat
Because at Horus there are 2 free trimm switches  in the middle 
this is the same as Toggel Switch  (ON)-OFF-(ON) option with autorepeat

think this functions are ready for trimm switches

fraternl, 2015-12-31:
>> And no, the companion simu corrently doesn't take any of the switch settings into account.
I don't believe there's a request for that.
Would I need to do that or can I use this ticket for it?

hrenz, 2015-12-31:
Hy 
the tree pos switch it self are integrated, thats ok, and enough for  (ON)-OFF-(ON) too
I think what realy would be interesting is a autorepeat funktion like the trimm switches has.

mhotar, 2015-12-31:
I would see the SF AdjustGV +x/-x with autorepeat function (once, repeat every x second), similar to like PlayValue has, sufficient to create this type of switch. There is opened issue for this already somewhere.

hrenz, 2016-01-01:
Hy,

think autorepeat funktion like the trimmswitches we have would be right way for  (ON)-OFF-(ON)

projectkk2glider, 2016-01-04:
I have this idea for Companion simulator:

Add new tab to the radio profile (for all Taranis variants) where user can define for each switch (that the radio supports):
 * switch physical position (important especially for X9E)
 * switch type (more precise than in the radio settings, like three pos momentary), can also be none if the switch is not installed.

This new setting will then be used to populate the switches in the simulator window. 

The related issues is: #2377

kilrah, 2016-01-05:
Yep why not, maybe with a way to "import current radio settings" so that you don't have to set up everything twice completely, only once on the radio, then import and edit the differences.

lshems, 2016-01-07:
Made a small screen mockup. The idea would then be to assign the physical switch positions PA, PB ... to the named positions SA, SB, SC, and have the physical possibilites follow the physical switch positions.

All functions in the rest of the programming would refer only to the named positions.

This would also allow for an often asked feture to interchange switches easily,without looking at all possible places.

[image]

closed 2016-11-27 by fraternl