opentx/opentx #3190: Turning off "trim sounds"

fraternl, 2015-12-31:
It is possible to use the trim buttons for something else, sometimes inside a LUA-script (for instance changing a music track).
At those moments you don't want the trims to influence the flight controls.
As a trick you can create a function to change a dummy GVAR.

At such times one doesn't want to hear the "trim center", "trim maximum" and the beeps as they don't make sense to what you are doing.
Maybe turning them off always when they are not actually used for flight is sufficient, but that might be hard to implement.

kilrah, 2015-12-31:
You can turn the desired trims off in the flight modes page.

fraternl, 2016-01-01:
I didn't think of that...

I checked it just now and you are right.
The trim is then turned off and no sound are heard, but if I would have my trim set to some value this would be turned off.

For my purpose I would first have to copy my trim values to subtrim before switching to this other flight mode.
So it's not an out of the box solution.

Do we have an SF for that?

fraternl, 2016-01-01:
Turning of these sounds when a trim is used to set a gvar would make things a lot less complicated.

Thinking about it. It's rather handy to change flight modes when you don't really want to change flight behaviour.
A flight mode can only be used for models that aren't using flight modes for its normal uses.

I think it's quite elegant to turn these sounds off during this SF.
Especially when the variant "+/- increment" is used.

fraternl, 2016-01-01:
@Kilrah

Could you please reopen this ticket...
On second thought your proposed workaround is not a practical solution for my problem.

kilrah, 2016-01-01:
> Turning of these sounds when a trim is used to set a gvar would make things a lot less complicated.

This was already debated and most considered sound to be beneficial when adjusting a GVAR as it tells you when you reached a point where further presses don't cause anymore change, and even more importantly warns you if you're about to go negative from positive or vice versa which could be dangerous. And even for cases when the sounds are not useful it's not the end of the world to have them, is it?

fraternl, 2016-01-02:
Compared to the end of the world all other stuff is always unimportant :-)

The Taranis is not made for playing music, I know this. For most people it is probably a silly thing to do. But for many people our hobby itself is silly. I replaced the speaker of my Taranis, opened up the grill, modified the PCB to give the amplifier 5 volts instead of 3.3 volts. The speaker compartment has 2 leds. One for power on (which also lights up when it's connected to USB) and the other one for charging.
Too geeky maybe. 

I like it that one can create a jukebox with it. And after taking it to this limit this is the one thing that's left.

I'm just looking for a way to turn these sounds off. It would be great if this last hurdle can be taken.

A LUA function to enable/disable a flag would be enough.

lshems, 2016-01-02:
There is another work around: allowing the normale buttons to trigger an
event in telemtry script background. Then you don't need to program
anything special, an use the same control buttons as in run mode.

This is in my opinion much nicer and more elegant.

The script needs to decide whether you do something with it.

The event info should be passed to the last script that was active in
screen. That is, the script that will reappear on screen when reentering
telemetry screen after exit.

fraternl, 2016-01-02:
I don't know exactly what you mean by this.
The buttons are already used to trigger a LUA-script.

The problem is that this doesn't deactivate the trims nor does it turn off these sounds.

Maybe the right solution should be that the trim function (and sounds) should be turned off when a button is set to a LUA-Script.

kilrah, 2016-01-02:
> The Taranis is not made for playing music, I know this. For most people it is probably a silly thing to do. But for many people our hobby itself is silly. I replaced the speaker of my Taranis, [...]

If only people who like doing that kind of things and spending the time to do these builds/mods would also take the time to tackle the software side of it instead of asking us, everything would be so easier and constructive :)

I'll reopen the issue in case someone ever wants to handle it. 

>There is another work around: allowing the normale buttons to trigger an event in telemtry script background.

I would absolutely not want flight-critical switches to have to go through the Lua environment. What you describe would mean when you press the trim keys the even would first get sent to Lua to see if someone wants to do something with the keypress before it can do its intended job.

lshems, 2016-01-02:
As I understood, you want to scroll through music songs while flying. You
do so by adjusting a gvar on trim button click, but the trim sounds are
annoying.

The script that plays your songs could check for the plus or minus button
instead. But currently you cannot check when the script us not in run mode
(shown on screen)..

I now use stick control to scroll up and down through options, others have
a (on)-off-(on) switch and check that.

If I have not understood your usecase, then could you try to make a story
like: when playing music I would like to be able to select the previous or
next song easily.

That would perhaps trigger some other suitable solution, or make clear this
is really needed on trims.

Guido

lshems, 2016-01-02:
I would absolutely not want flight-critical switches to have to go through
the Lua environment. What you describe would mean when you press the trim
keys the even would first get sent to Lua to see if someone wants to do
something with the keypress before it can do its intended job.

That is not the intended use case.

I see no problem in sending the plus, minus and enter events to the lua
environment from the default screens. They respond only to page and page
long, menu and menu long.

the channel monitor screen is the only one that uses exit to return to the
first default screens, and that seems not logical to me. If that response
would not be there, the entire navigation would be logical:

menu to change the active menu, page to change pages within an active menu,
plus or minus to change an item on a menu page.

enter to select and deselect an item for edit, plus and minus to change the
value to edit.

exit is a little bit obsolete in this scenario, but I would like it to
return to the previous menu.

personal preferences:
I don't like the pages within a menu to change on plus and minus, it is
confusing to me. We already have the page button for that.

I don't like the exit on the channel monitor screen, it is not logical,
since the other main screens don't respond to it.

i don't like the change to "long exit" to exit a telemetry screen, it was
better in 2.0, where it would return to the last menu on "exit" only.
people will get stuck not nowing to press "long enter".

hope this is helpful,

guido

Met vriendelijke groet

Guido ter Horst
+32 499 708000

fraternl, 2016-01-02:
No, that's not how it works.

Using a SF I have set the rudder trim to run a LUA-Script.
This LUA-script can change the music track

When you enable this function this doesn't disable the normal trim function.

What I did is turning off the trim by setting rudder trim to set a gvar when music is played. This is almost the workaround I want, but it still gives me these sounds. I also lose a gvar, which I really don't want.

The only thing I want is turning off the "mutation of the trim value" not turning off trim.
If I would turn trim off it would influence flight if trim is not set to neutral.

@kilrah commented about letting flight critical stuff go through the LUA-environment.
That's not the case. A special function is already triggering a LUA.

I don't think anyone thought of it if the normal function should continue. Currently the trim does 2 things.. LUA and trim.
I'm proposing to turn off the trim when a SF sets a trim to a LUA-script

lshems, 2016-01-02:
Thanks Fraternl,

that clears it up.

The nice thing in your setup is that it doesn't need a telemetry script running.

but if the plus and minus buttons would be available in LUA as an event during display of the default page, converting your script to a telemetry script could handle your usecase.

it would make the SF and GVAR free for other use again.

I think I will post the menu adaptations and event passing as two new issues. They are useful, and I can use them myself as well, since they support a general usecase of using some buttons in the background mode for the "seelcted" telemetry script, to do some actions.

fraternl, 2016-01-10:
"Turn off trim mutation during the time that a trim button is assigned to a LUA-script."

sande005, 2016-04-22:
While having a notice of some kind for trim center is generally fine, I have a current nitro set so that trim only effects the very low end.  Full down trim closes the carb, full up trim opens it to a very fast idle.  That way I can tweak it while it is running for the best idle, which sometimes changes depending on weather.  As I pass through trim center, there seems to be a pause as the radio plays the wav file (but it could just be me imagining).   Not the end-of-life-on-earth, but if there was an option to enable/disable, even better individually, that would be nice, as in that instance I don't care if it is centered or not so the prompt is just distracting.

kilrah, 2016-04-22:
I believe if you enable the "trim idle only" feature that's precisely there for that case there is no more center.

sande005, 2016-04-22:
True, but I'm using cross trims, so would have to disable it on the ele trim!

projectkk2glider, 2016-04-22:
You have (re)discovered the issue #170

closed 2016-11-27 by fraternl