opentx/opentx #3191: opentx doesn't accept this radio protocol

powercycle, 2016-01-01:
OpenTx 2.1.6
Taranas Plus
Taranis BootLoader - 2.0.9

When saving the epee file using the Taranis plus radio profile, the error:  OpenTX doesn't accept this radio protocol error occurs.

The expectation is that it will save the file without error.

I'm happy to try things to clear this error, however, it is not obvious.

[image]

powercycle, 2016-01-01:
This occurred after opening an eepe file created from a 9XR radio profile.

Saving a new empty file works as expected.

kilrah, 2016-01-01:
What version was the file created by?
Are you sure you then set companion back to Taranis Plus before saving?

Can you upload the file to a file sharing site and post a link?

powercycle, 2016-01-01:
I was able to resolve this by creating a new file and then copying the models to it.  It then saved without error.

The file from the 9XR was created with version 2.0.14 that I built on my Linux machine.

However, I tried to re-create the issue and was unable to do so.

We should probably consider this resolved unless you would like me to try something further.

[opentx.eepe.zip](https://github.com/opentx/opentx/files/76041/opentx.eepe.zip)

kilrah, 2016-01-02:
The problem is due to the 9XR having module protocol set for the to PPM (normal), but when transferred to Taranis it is not a valid option for the internal module.

You must manually change the protocol in each model to the appropriate one for the receiver you have in that model.

closed 2016-01-02 by kilrah