opentx/opentx #3192: How to: Update your XJT Module

wot4, 2016-01-01:
There is many different instructions available out there and some are more explained than others on how to update your XJT firmware on the Taranis.  Below is the step by step for updating the internal XJT and the external RX's. Some say you need to buy an external USB converter cable (http://www.frsky-rc.com/download/view.php?sort=&down=174&file=How-to%20built-in%20XJT%20upgrade) to update the internal XJT module, but after many hours of research, below is what I did and has worked.

This process worked for a Taranis X9D Plus.  My old firmware on the Taranis was Version 2.0.13, I updated this to Version 2.1.6 (http://www.open-tx.org/2015/11/11/opentx-2.1.6/).

As of 01/01/2016
I updated my internal XJTmodule and the RX to Version 151223 (the latest EU version) (http://www.frsky-rc.com/download/view.php?sort=&down=217&file=EU%20LBT%20Firmware)(The XJT and RX download are the same because after you have downloaded the .rar file it has the different .frk files for the different products) 

Below is the step by step for what I did for updating the internal XJT and the external RX's:

	1	Connect the Taranis to a computer.  You should see two drives mounted, one is called Taranis and the other is the internal SD card. 
	2	On the internal SD card under a directory called "Firmwares" copy the downloaded version file (.frk) into it.   
	3	Safely remove (or unmount) the drives and turn off the Taranis. 
	4	If you're flashing an external device plug the unit via a cable into the pins of the module bay on the back of the taranis. (Using a 3 pin custom cable, THIS IS NOT A STANDARD SERVO CABLE) 

[image]

	5	Turn on the Taranis. 
	6	Long press the Menu button to get into the Setup menu.  Press Page until you get to the SD Card browser. 
	7	Navigate to the Firmwares folder and press Enter. 
	8	Select the firmware file you copied onto the SD card in step 2, and press Enter. 
	9	Select what you want to do, Flash External Device, and press Enter, or if you are updating the internal XJT select Flash Internal Device. 
	10	A progress bar appears, and when it's completed the device has been flashed. 

Hope this helps anyone else who has also been looking into this.

wot4, 2016-01-01:
Word of Warning:

The voltage pin in the module bay puts out the TX battery voltage.  If you're using a non-standard battery, e.g. a 3S LiPo, then make sure the device your flashing can handle this voltage !!

griebd, 2016-01-01:
Just a question: is there any way to know which firmware version is already loaded on the XJT or on the RX?

kilrah, 2016-01-02:
Hmm the issue tracker is not for documentation, this will just get closed and lost. It would be best if you could transfer that into a wiki page:
https://github.com/opentx/opentx/wiki

> is there any way to know which firmware version is already loaded on the XJT or on the RX?

No.

closed 2016-01-02 by kilrah