opentx/opentx #3196: Thottle and Aileron switchs stuck since upgrade to 2.1.6 on E9XR

momohelico, 2016-01-02:
Hello,
I've upgrated my turnigy 9XR with open TX v2.1.6 and during the update it display once EEPROM corrupted on the transmitter screen but all is fine for the companion9X software.
After upgraded, transmitter works fine but THR and AIL swithc don'y work anymore !
Is someone has a solution ?
thanks

PS: i downloaded several time the file "opentx-9xr128-frsky-fr-2.1.6.hex" and performed several time the upgrade but nothing changes...

kilrah, 2016-01-02:
The "frsky" opttion must only be ticked if you made the hardware modification to have telemetry on your radio. 
Unselect, redownload and reflash.

closed 2016-01-02 by kilrah

momohelico, 2016-01-02:
THANK YOU VERY MUCH, it works... (sorry, i didn't know, i thought that Frsky box was for a Frsky module)