opentx/opentx #3200: New 9XExtreme Board OpenTX support

ckleanth, 2016-01-04:
Steven has a new addon board for the 9x. Currently its supported by MIke and Er9x. Can we add OpenTX support please?

thanks.

bsongis, 2016-01-04:
I am not sure any OpenTX dev has the board right now!

ckleanth, 2016-01-05:
I will ask Steven

bcsteeve, 2016-01-11:
Hi.  Steve from SmartieParts here.  I've had a dozens of people (and that may easily be an understatement) asking for OpenTX support for this item.  I tell them all to bug the devs... looks like one finally did so :) 

I'd be happy to donate a unit to the cause, with only the request that it not be for nothing.  I don't mind giving out a few to multiple devs with the modest condition that if it isn't supported within a reasonable time that they be sent back.  (ie. my concern is sending them out with no real effort put into supporting the platform)

bsongis, 2016-01-12:
@bcsteeve Hi Steve, I will do the port myself if nobody wants to take it, but I have too much work right now, I can do this after the Horus port is done. Nobody else wants to help on this one? It's not a too much complex job, as Mike already wrote the low layer code, it just needs to be glued to OpenTX. I would assist the volonteer as I did with Franck of course.

DieHertz, 2016-01-13:
@bsongis I have a 9x, 9xtreme, programming background and desire to help, but I'm not familiar with OpenTX nor ersky9x code.
I've had a look at OpenTX code for other STM ARM based boards, but there are so many magic constants like port numbers (I guess) and such I'm afraid I won't make any sense of it without knowing 9xtreme internals :-)

bsongis, 2016-01-13:
You could try to create a new branch, add a new board in the targets directory and start porting all drivers from ersky9x there. Perhaps in your place I would first read the ersky9x code for the 9xtreme to see if I would be able to do the job!

bcsteeve, 2016-02-25:
Just thought I'd poke my nose back in here.  I get requests almost daily for OpenTX support for the 9Xtreme.  I keep telling them the same thing... I'm not the developer, and there has to be community support.  I tell them to bug you guys :)  I'm not sure if that's happening.  

@bsongis if you want to tackle the port and need a unit, just let me know and I'll get one out to you.  Same for any other active dev willing and able to do this.

Cheers.

jeremyroe, 2016-02-25:
Another 9xtreme owner here bugging devs for OpenTX support.  I dabble in coding so I'll look things over and see if I can contribute also.  Sounds like Steven is open to providing dev boards to those interested and I'm willing to run unstable code and be a guinea pig also.

projectkk2glider, 2016-02-25:
The problem right now is that all new development is focused on the Horus. I can not imagine that any of the current developers (there are basically only three of us) will drop the Hours and start porting this board. We need new blood.

jeremyroe, 2016-02-25:
I understand - I'm trying to dig into it now and reading through docs - if I can I'll try to pitch in myself.  My background is in basic (years ago) some vbscript and most heavily in javascript and sql so there will be learning curve but I'm up to try.

bsongis, 2016-02-26:
@bcsteeve I have too much work right now.
I have 3 new ports (Horus is one) for OpenTX in progress for big LCD screens and it's too much. Plus new telemetry features coming for TBS (Crossfire protocol) and another protocol for Dronecode.
This port is not so difficult, is it possible somebody else owning the board gives a try? We really need volonteers, it's becoming a non paid full time job ...

bcsteeve, 2016-02-26:
Of course, I understand.  And as Jeremy said, he's going to take a shot at it.  I have no firmware coding skills at all, so I'm no use in doing the port.

When you guys decided to go from the 9X to basically anything... yeah, you took on a lot.  Where does it stop?  Take care of yourself!

flexd, 2017-05-03:
Did anyone get around to porting opentx to the 9xtreme boards? I can't see any issues/commits/anything related to it, so I think I already know the answer.

schwabe, 2017-05-03:
I think the point is still the same, no OpenTX dev has one of the boards. Considering that the 9xtreme board has a STM32F205 processor and we now have the X7, the port might now be even less troublesome. But at the moment there so many things going, the motivation for most us to invest time into this is probably not that high.

kilrah, 2017-05-03:
Yes the large amount of work that was needed was made for the X7, so it should be relatively straightforward... but then since the 9Xtreme was discontinued I believe 6 months to a year ago there's little interest in investing effort in it.

We'd be happy to merge in the code if an actual user of this board did the port.

flexd, 2017-05-12:
@kilrah I do not know what is needed, and I have little practical experience with C, but if someone is willing to at least outline the things that need to be done I can at least take a look at it. I have a 9X with the 9xtreme board in it, which is why I was asking in the first place :)

kilrah, 2017-05-12:
If you look in the radio/src/targets directory you'll see different subdirectories containing the hardware definitions and drivers for each platform. That needs to be created for this platform. Given it uses the same processor family as the Taranis some stuff could be reused from there but most likely needing modifications. Some stuff could be taken from ersky9x too.

closed 2017-11-12 by kilrah