opentx/opentx #3203: TRx not working on 9XR pro

estechnical, 2016-01-04:
Hi, I'm using OpenTX 2.1.6 on Turnigy 9XR pro and I am having trouble getting the PPM in working as I wish...

I have 16ch radio module (XJT) and I'm using some channels to control a gimbal and camera, I'd like the trainer port to allow me to have a second TX for these tasks.

I have so far found that only the first 4 channels of the incoming PPM are read correctly, coming from my 6ch second TX. All the other TRx inputs all give mid-point values.

Having looked at some of the other issues related to the trainer port I changed to PPM out instead of XJT and the PPM inputs all started working as expected - but of course now I can't use my radio to it's full potential...

Is this something that's not possible for my case?

Thanks,
Ed

estechnical, 2016-01-04:
Further to my observations above, if I start up the radio having previously set it to XJT mode the PPM in doesn't work (as per above) but then if I switch to PPM mode and back to XJT mode the PPM (TRx) inputs all remain working, including those above ch4. This leads me to think I found a bug and I'm not doing something wrong... Any ideas?

Thanks,
Ed

projectkk2glider, 2016-01-04:
Definitely looks like a bug.

projectkk2glider, 2016-01-04:
@estechnical Did it work in any previous 2.1.x version?

estechnical, 2016-01-04:
How would I get earlier firmwares?

estechnical, 2016-01-04:
Well I tried to grab an older firmware but it appears to be hard to get old firmwares (probably deliberately?) - I can't comment on older firmwares as I can't test and don't know.

Sorry... if there is anything you need me to test please let me know.

estechnical, 2016-01-05:
The issue persists in 2.1.7, I just tested and the behaviour is the same...

projectkk2glider, 2016-01-05:
>How would I get earlier firmwares?

You can't unless you have the old firmware stored on your computer. Anyway I just wondered if any of previous versions worked (which would then mean that we messed up something) or it was newer actually implemented.

estechnical, 2016-04-27:
Hi, has there been any progress on this issue? I'm happy to test if needed. I need this issue fixed as soon as possible because it's really hampering my use of the radio...

projectkk2glider, 2016-04-27:
I will take a quick look at the code if something obvious is missing. I don't have this radio, so it is impossible for me to test...

projectkk2glider, 2016-04-27:
Sorry, I have found nothing obviously wrong. This must be tackled by some other developer.

bsongis, 2016-04-27:
I take it

estechnical, 2016-04-27:
:) Thanks Bertrand

bsongis, 2016-04-27:
Do you know if it worked in 2.0?

estechnical, 2016-04-28:
Sorry but I never used this version. Is it possible for me to get this firmware to test for you?

projectkk2glider, 2016-04-28:
Download Companion 2.0.19 http://www.open-tx.org/2016/01/15/opentx-2.0.19 and with it the 2.0.19 firmware for your radio. Be sure to make a backup of your current setup as the radio/model settings will be wiped after you install 2.0.x OpenTX on the radio!

bsongis, 2016-04-28:
@estechnical Would you send me an email so that you can test for me?

bsongis, 2016-04-28:
I am testing on mine first, it was taking dust ...

bsongis, 2016-04-28:
It works here

closed 2016-04-28 by bsongis

estechnical, 2016-05-02:
Many thanks Bertrand, i think the reason was partly because of a dodgy contact in one of my plugs. Sometimes I was able to get the radio to vibrate when moving the cable... i have now pulled out a few contacts from the affected plug and swapped in another pin. This has helped with the issue, i no longer have to switch modes for the radio module to get trainer ppm in working. Perhaps your changes will make it more robust than before when incoming ppm signal is intermittent during plugging in the cable or in use. :-)

commit 9106ee58 (2016-04-28): [Sky9x] Fixes in Sky9x pulses driver. Would if fix #3203 ?
commit 2c08bb1b (2016-04-28): Fixes #3203 - Tested