opentx/opentx #3205: OpenTX 2.1.18 Avast antivirus false alarm virus alert/thread blocked

ckleanth, 2016-01-05:
OpenTX 2.1.18 Avast antivirus false alarm virus alert/thread blocked
affected files
C:\Program Files (x86)\OpenTX\Companion 2.1\opentx-9xrpro-simulator.dll
C:\Program Files (x86)\OpenTX\Companion 2.1\opentx-ar9x-simulator.dll

win32:Evo-gen [susp]
PID process 5800

projectkk2glider, 2016-01-05:
I guess false positive. Did it happen with previous versions. Try to upload those files to virustotal and see what you get.

projectkk2glider, 2016-01-05:
Companion 2.1.7 installer https://www.virustotal.com/en/file/7b2b8cf680490030882a48d1862b80feb7c692390ebb44882acf4bf5fd0f4e8f/analysis/1452015051/

2.1.6 https://www.virustotal.com/en/file/542df3a89c5b8e31056546cb8fbedc1cd2a5163258c94f183ea40fb60f6251dc/analysis/1452015366/

I say we are OK. Only 3 out of 53 anti-virus scanners think something is suspicious.

projectkk2glider, 2016-01-05:
OTOH the 2.1.5 is totally clean https://www.virustotal.com/en/url/e7039d65f8ee6f0c26c32d10142e1c2ddaf882ca4bbb1050f7c80cd579c2223b/analysis/1452015798/

@opentx/commiters  Did we made any changes to the 9XR-PRO and ARX9 that would cause that?

projectkk2glider, 2016-01-05:
Interesting: Nightly scans:
* N569 1 antivirus scanners complain
* N570 2 antivirus scanners complain
* N572 3 antivirus scanners complain

ckleanth, 2016-01-05:
This was after updated to 2.1.18. All previous versions were fine on the same machine. (Btw windows 7 x64)

mpjf01, 2016-01-05:
In my case Avast prevented installation of Companion until I temporarily disabled it. After successful installation I now get the warning when I run Companion. Asked for help on this a couple of days ago on OpenRC Forum.

fraternl, 2016-01-05:
Get a decent AV like Eset's Nod32
All free AV's are crap anyway, just slowing your system down and nagging with false positives like this one. While it is hogging you down it lets real viruses do their nasty work.
Just install their free trial and it will probably find some viruses like on most systems I get asked to fix.

projectkk2glider, 2016-01-06:
Interesting: 

Companion 2.1.7 compiled on my Windows XP VM in Debug mode shows no problems https://www.virustotal.com/en/file/10e226c9f5a57f5106b5a26545442bd7a9f8e9fa46de97784538af6f5031ed6a/analysis/1452117665/

Release build of the same sources produces 2 warnings https://www.virustotal.com/en/file/d868992c5784966170923c4478d05ec98418aaadaa4bf1d4a482df9af78febb3/analysis/1452117991/

So it is looking more and more like a false positive.

closed 2019-08-17 by kilrah