opentx/opentx #3206: Timer3

PieroCo, 2016-01-06:
Timer3 (persistent) in statistic is correct but in telemetry page is wrong.

[image]

Opentx 2.1.6 and 2.1.7

projectkk2glider, 2016-01-06:
The "bug" is in the TM3 name which is actually a global timer. this one was in OpenTX for a long time and was named TM3 since we had only two model timers. Now that timer 3 was added, the name TM3 should be changed.

So Timer3 and TM3 are different timers! To reset TM3 hold ENT in STATS screen (I think)

bsongis, 2016-01-06:
I see 2 bugs:
- [ ] In statistics a timer > 60 minutes should be displayed hh:mm:ss, otherwise mm:ss
- [ ] The same in telemetry screen

bsongis, 2016-01-06:
@projectkk2glider would you check my fix in 'next'? I will backport it to 'master' if you like it

projectkk2glider, 2016-01-07:
Looks perfect :+1: 
[image]
[image]

commit bbe4e7d4 (2016-01-06): Fixes #3206
commit 9fb7f120 (2016-01-07): Fixes #3206
commit ed9a8c99 (2016-01-07): Fixes #3206 - I forgot to add the file after the mods
closed 2016-01-07 by bsongis