opentx/opentx #3207: Telemetry Simulator

PieroCo, 2016-01-06:
the altitude is not populated  opentx 2.1.7
[image]

axyspg, 2016-01-06:
Not sure what's happening with instances with this version, but setting it to 32 fixes the problem.
I've read somewhere that 28 is the max instance in SmartPort, is that true? if yes, why works when setting instace to 32?
Let me congratulate all of you for this feature, it's really nice for LUA scripts development.

axyspg, 2016-01-06:
By the way, telemetry simulator doesn't send data when values are at 0. Is it the intended behaviour?

jtaylor2, 2016-01-06:
@axyspg 
>By the way, telemetry simulator doesn't send data when values are at 0. Is it the intended behaviour?

Yes, it is intended behavior.

kilrah, 2016-01-07:
I haven't followed the simulator enhancements closely, but before that the Instance number was the "encoded" one, not the "user" one i.e. not 1, 2, 3, 4 but 0x00, 0xA1, 0x22, 0x83.

I'd expect that would have been changed in the refactoring, but maybe not. And yes it sends only with the chosen instance, that's the whole point of it.

pointu, 2016-01-10:
The VSpd value is not populated too,
Can u save the last used values ?

projectkk2glider, 2016-01-11:
>The VSpd value is not populated too,

My pull #3207 should cover this

closed 2016-01-11 by dsbeach

dsbeach, 2016-01-11:
The best way to save values is via the "Load" function of "Replay SD Log File".  There are currently no provisions for just retaining the last used values.  Note that if you are using the telemetry sim to develop Lua scripts you can just change your script source and press 'PF7' to test the changes without exiting the sim.

PR #3213 (2016-01-11): Fixes #3207: proper sensor instance generation in Telemetry Simulator
Closes #3207
commit b4937544 (2016-01-11): Fixes #3207: proper sensor instance generation in Telemetry Simulator
commit f8de9f9d (2016-01-11): Merge pull request #3213 from opentx/projectkk2glider/issue_3207_telem_simu_instance

Fixes #3207: proper sensor instance generation in Telemetry Simulator
commit a3fac604 (2016-01-11): Fixes #3207: proper sensor instance generation in Telemetry Simulator (ported from master #3213)