opentx/opentx #3208: Trainer slave output on 9XR-PRO

DanNixon, 2016-01-06:
When I try to use my 9XR-PRO with Phoneix flight simulator via the JR port all I can ever manage is for Phoenix to detect a PPM signal (the transmitter not detected message disappears) however all channels read zero regardless of the actual channel values on the 9XR.

I have the following configuration:
External RF: PPM, channels 1-8, 22.5ms, 300us, negative
RF Port 2 (PPM): channels 1-8, 22.5ms, 300us, negative

The same settings work fine when I use the factory ersky9x firmware.

projectkk2glider, 2016-01-07:
Which OpenTX version?

DanNixon, 2016-01-07:
2.1.7

DanNixon, 2016-01-09:
Just an update on this, I had a look at the trainer PPM output of my 9XR-PRO and it is actually outputting a valid signal.
The difference is that the 9XR outputs an active high signal whereas the Taranis outputs an active low signal, this seems to be the difference in getting simulators to work.

kilrah, 2016-01-09:
Then the 9XR's PPM polarity is probably inverted with regard to the display, choosing positive should do the trick.

DanNixon, 2016-01-09:
I believe it is, on both the module output and trainer port I get a active high signal when the display indicates `-`, this is the opposite of what happens on the Taranis.

(I also noticed #1447 which seems to be very similar)

ikarus23, 2020-04-07:
> Then the 9XR's PPM polarity is probably inverted with regard to the display, choosing positive should do the trick. If you confirm we can correct it.

Can confirm! Changing the polarity worked.