opentx/opentx #3209: telemetry screens navigation: just push exit instead of long push exit to exit

lshems, 2016-01-06:
i don't like the change to "long exit" to exit a telemetry screen, it was
better in 2.0, where it would return to the last menu on "exit" only.
people will get stuck not knowing to press "long enter". 'panic button' function for 'exit' is lost.

projectkk2glider, 2016-01-07:
Short EXIT is now available to Lua Telemetry scripts, see #3112. We can't have it both ways. It is either available/seen by Lua or not.

lshems, 2016-01-07:
OK. That's due to the LCD.lock() that is gone, I assume.

When I create custom menus, I need the short exit to have the behaviour
identical to the native menus, so that's fine.

Would it be possible to force or send a long exit key press to the event
handler to force a telemetry script exit?

Then all is fine, and I can handle the short exit myself and use it to
close my menu when no edits are active in my menu.

projectkk2glider, 2016-01-07:
>Would it be possible to force or send a long exit key press to the event handler to force a telemetry script exit? 

Currently no, but this could be an useful addition to Lua, we just need to come up with a sensible interface for it.

lshems, 2016-01-07:
how about a new function telemetryscript.background()

jcowgar, 2016-01-14:
Please keep Short Exit available to LUA scripts. I use it in my Lap Timer script

https://github.com/jcowgar/opentx-laptimer/issues

https://youtu.be/mqHtRXctdZs?t=5m38s

lshems, 2016-01-14:
As I understood that is the new standard. Nothing to worry.

I just need to be able to detect a short exit, (can already be done, so
fine) and then be able to quit the telemetry pages (LCD.exit()????).

het volgende:

projectkk2glider, 2017-03-21:
We decided that the script will be able to request the exit from the telemetry screens back to the main view.  How exactly will this be implemented is not yet decided, options are:
 * new function
 * return value from foreground function (current favorite)

closed 2020-04-06 by stale[bot]