opentx/opentx #3210: pass the plus, minus and enter events to LUA when in default screens

lshems, 2016-01-06:
the LUA telemetry scripts are always running in the background. Only when you enter the telemetry pages, they will allow for some screen output. Currently, you can only "capture" the events on the display controls from within a LUA script when on the telemetry pages. Since the plus, minus and enter buttons are nit used in the default pages, why not send them to the lua environment, even in the background, to the last selected telemetry script.

Use case: I am using te telemetry scripts for running a special competion F3K training program, controlling timers, flight logs and graphs, and more. 
It would be nice that if I exit to the default screens, I could use the same controls, being the up, down and enter buttons, to change the current task and perform other actions, such as signalling a false landing or skipping a task or time.

I don't need to see the screen to be be to use the right hand side screen buttons.

projectkk2glider, 2016-01-07:
Why can't you stay in your telemetry view? You say you don't look at the LCD so it does not matter what is displayed. 

The whole concept of keys controlling something that isn't "active" sounds unintuitive to me. Much better for display to be in sync with the key actions. So when you see custom Lua screen, you can control it with keys, when you don't see it you can't. Isn't that what a normal user would expect?

lshems, 2016-01-07:
It is just an opportunity to make the buttons useful, that otherwise remain
unused.

Nice thing of opentx is that the use cases don't always need to be clear
upfront :).

No prio, just nice to have, and would solve the trim sounds problem for
people using trims to change gvars on a logical switch. Use a telemetry
script to do that.

P.s. my requiremt on focus only on the 'active' script is not neccesary. If
you use it, make sure yourself there is no overlap between scripts waiting
for the event.

jtaylor2, 2016-01-07:
>So when you see custom Lua screen, you can control it with keys, when you don't see it you can't. Isn't that what a normal user would expect?

That is certainly what I would expect.  Having input keys controlling a background task just sounds confusing and somewhat dangerous.

>No prio, just nice to have, and would solve the trim sounds problem for people using trims to change gvars on a logical switch. Use a telemetry script to do that.

And having an input key (as opposed to a flight mode control) controlling a flight mode parameter from a non visible telemetry script sounds even more dangerous.

dsbeach, 2016-01-07:
Agreed - never have keys influence something you can't see.  However, users are accustomed to using switches do that sort of thing. 

Why not use Lua mix scripts to control telemetry scripts when you can't see them?  That is my plan for Lua GPS triangle racing.  Both mix scripts and telemetry can share global variables, and using a mix script with a 'SOURCE' input allows users to select their preferred switch (see https://opentx.gitbooks.io/opentx-lua-reference-guide/content/input_table_syntax.html).

I would also recommend audio feedback via playFile() when the mix is 'triggered'.

lshems, 2016-01-07:
we can close it as far as I'm concerned. I don't really need it (yet).

But I think you need to try and think in possibilities, not in impossibilities. It would be usefull as user input for for instance changing volume with plus and minus, instead of using a flight control input as a knob or stick for doing that and actually loose inputs for flight control since you ave assigned a slider to volume control. 

I don't agree with the argumentation either: 

I see no reason why flight parameters can be altered using buttons when a telemetry screen is visisble, and why we would block it when not visible.

It is not that the telemetry screen will actually show what the buttons are doing in either case, so it really makes no difference to me.

You can as easily make a telemetry script that shows a text saying:"I will randomly change some settings on pressing the plus button" and have it execute something of which you have no clue what it will change. It could  delete all your inputs when you press the plus button, without the user knowing. 

The first virusses are there to come. LOL

projectkk2glider, 2016-01-07:
> You can as easily make a telemetry script that shows a text saying:"I will randomly change some settings on pressing the plus button" and have it execute something of which you have no clue what it [...]

You could certainly do that, but that would obviously be bad practice. I will open an issue for Lua anti-virus now... :stuck_out_tongue_winking_eye:

lshems, 2016-01-07:
LOL

closed 2019-11-15 by stale[bot]