opentx/opentx #3211: Companion missing new context menu on Flight Modes

elecpower, 2016-01-07:
Taranis X9D tested on Win 10 & Ubuntu 14.4
Companion 2.1.7
Release notes "Added Clear option to the flight mode context menu (right click on the Name label) (#3026)"

projectkk2glider, 2016-01-07:
Perhaps the explanation in release notes is not clear enough, click on the "Name" text that is on the right side of name input control:
[image]

closed 2016-01-07 by projectkk2glider

elecpower, 2016-01-07:
Thanks. The flight mode tab was a more obvious choice since the whole flight mode is what you are clearing rather than the Name field. 
But now have found another problem in that Clear does not reset every field of the flight mode with the main one being the switch thus it is still active. I'll raise a new issue since the title will be different.

projectkk2glider, 2016-01-07:
Qt does not support context menu on tab unfortunately, so the name label was chosen.

elecpower, 2016-01-07:
Guessed it was a technical limitation of the GUI. Maybe QT v5 permits which I notice some work has started on moving to for v2.2