opentx/opentx #3214: Companion flight mode Clear does not reset all fields to defaults

elecpower, 2016-01-07:
v2.1.7 tested Win 10 & Ubuntu 14.4

For flight mode 0
- GVARs names are not cleared
- GVARs Popup enabled are not unchecked
For flight modes 1 thru 8
- Switch is not reset to dashes thus the flight mode is still active
- GVARs source not reset to "Flight mode 0 value"

elecpower, 2016-02-10:
Performed some more testing to discover slightly different behaviour to reported above for clearing FM 1 - 8:
The Switch is reset to dashes but the screen field is not refreshed after clear.
All GVars are set to Own Value which is incorrect and the screen combo boxes are not refreshed.
Sets the Trims to Use Trims for Flight Mode 0  which is correct behaviour however it triggers zeroising the Trim Values for FM 0 which is not what should occur.
The update but no screen refresh issue was found by performing a FM Clear, saving the model file and reopening.

elecpower, 2016-05-04:
I have written a fix which I would like to contribute. Developed in Ubuntu 14.04 with Companion 2.1 tool chain and local git clone of master and next. Have only tested against master 2.1 since you haven't published the 2.2 tool chain and which from other posts will be provided using docker.
If okay to contribute would the steps be for me be to self assign, create a issue branch from master, clone locally, apply the fix, merge, create pull fixed request. Please advise.
In testing I discovered an app freeze bug in the Rotary Encoders linked FM which I will raise as a new Issue.

kilrah, 2016-05-04:
Yep sure.
See here for a little guide:
https://github.com/opentx/opentx/wiki/Github-issue-workflow

If you don't have write access to the OpenTX repo yet you can create your own fork on github, clone that (or add as additional remote), create a branch in which you'll commit your fixes, push that to your own repo, then open a pull request from your repo to ours on the web interface. We'll review, request changes if needed, then merge.

closed 2016-05-08 by projectkk2glider

projectkk2glider, 2016-05-08:
@elecpower Thanks for your contribution. For `next` you don't need Docker containers. The build chain is mostly the same. 

On Ubuntu 14.04 install following packages: 
* Qt5 libs `qt5-default`
* clang `libclang-3.6-dev python-clang-3.6` (optionally)

I think this are the only new packages that are needed to compile 2.2 branch. There is also no makefiles anymore, everything is driven by cmake. Example to compile Companion:
```sh
#make a build dir in opentx dir
mkdir build-horus
cd build-horus
#run cmake
cmake -DCMAKE_BUILD_TYPE=Debug -DPCB=HORUS -DHELI=NO  -DDEBUG=YES -DCLI=YES  -DGVARS=ON -DALLOW_NIGHTLY_BUILDS=YES -DLUA=YES ../
#make companion
make companion22
#make radio firmware
make firmware
#make simu simulator (needs libfox)
make simu
```

More info https://gitter.im/opentx/opentx?at=571cf8220eef754b5ea9f2ce

elecpower, 2016-05-08:
@projectkk2glider thanks for the build chain info

projectkk2glider, 2016-05-09:
Please report back if my instructions were correct, perhaps we can then update the wiki page.

elecpower, 2016-05-11:
Had to also install qtmultimedia5-dev qttools5-dev qttools5-dev-tools to get Companion to compile.
Refered to radio/util/Dockerfile.
Haven't tried to compile firmware or simu yet but never had a need in 2.1. Though the Dockerfile if complete is perfect reference for those wanting to build outside Docker.

commit 91105a86 (2016-05-08): Merge pull request #3460 from elecpower/elecpower/Issue3214_flight_mode_clear

Fixes #3214 : Flight mode clear
commit b8b78a5f (2016-05-08): Fixes #3214: Flight mode clear
commit 5f648b20 (2016-05-08): Fixes #3214: Flight mode clear (ported from master, Re #3460)
commit 33bacd3e (2016-05-13): Fixes #3214: Flight mode clear (ported from master, Re #3460)