opentx/opentx #3216: timer "start" in lua not set to negative values properly

lshems, 2016-01-07:
in previous versions it was perfectly fine if you would enter a negative value in the setTimer function to the "start" parameter.

This now gives unpredictable results.

The value that is returned when reading the start, is dfferent also.

ive attached a test file and the outputs.

For me it is not a problem anymore, since I check the value before updating (lessons learned :) ) now whenever doing it in LUA.

It could be related to the problem 2984 that was reported, so thats why I post here.

[Timer.txt](https://github.com/opentx/opentx/files/81843/Timer.txt)
[image]
[image]

kilrah, 2016-01-09:
Negative timer start values have never been allowed, if that worked at some point it would have only been by chance.

lshems, 2016-01-09:
As said, I don't have a problem with it not being possible. If however the actual timer value is counting down into the negative, you perhaps want to set it starting counting from a negative start point as well.

Again, use cases become more complicated when using LUA.

But you can close it , noi problem.

projectkk2glider, 2016-01-09:
`start` parameter is internally unsigned and its a bug that Lua allows negative numbers there! Start can not be negative and can not be set negative in the radio. Only the timer value itself can become negative.

I will leave this issue open until we solve signed/unsigned issue in Lua API.

kilrah, 2016-01-09:
> If however the actual timer value is counting down into the negative, you perhaps want to set it starting counting from a negative start point as well.
Again, use cases become more complicated when using LUA.

A good reference is simply the radio menus. If you can't set something there (and you can't set a negative timer start value indeed), then you should not try to set one in any other way.

lshems, 2016-01-09:
Please be aware that we need negative times, to be able to measure flight
time on timers even after the target has been met.

If you intend to change something, it should be allowing negative timer
starts.

Thanks

kilrah, 2016-01-10:
> Please be aware that we need negative times, to be able to measure flight time on timers even after the target has been met.

As mentioned that is all good, the timer itself can be negative precisely for that reason. Not much sense allowing it for the start value, especially given that if a start value is set the timer will always count down and would just go further down in the negatives.

lshems, 2016-01-10:
Hi andré,

Your own explanation just makes clear why you want to set a negative start
value: it will just continue counting down from that point.

That means you can (re-) create a state at any moment which is perfectly
possible to happen, by setting a negative start value.

Or, put the other way around, if you can't set a negative start value, you
cannot (re-)create a state where you count down from a negative value,
other then waiting for that state to happen by time passing by.

This is very inconvienient when you want to be able to save a current
state, start something else, load the same state, and continue where you
left.

With lua being used as means of creating contest timers, this is a current,
realistic and feasable scenario: a forced break in the middle of a running
task, doing something else, and return to the task.

Just trying to explain the basic principle that in any programmable system,
you want to be able to create every system state as a start state. If you
can't, sooner or later you will run into a problem.

Guido.