opentx/opentx #3218: Localization request

Datospel, 2016-01-09:
Don't know whom to write to... I would like to contribute to localization of OpenTX for Russian/Ukrainian speaking audience. There are many members who are desperate to use OpenTX in Russian and Ukrainian languages.... I am ready to provide such localization... Whom to write to? Who is in charge of that question

Thank you
Alexander

bsongis, 2016-01-09:
It was the right place!
First you have to create a translations/ru.h.txt file and submit it in a pull request.
Here is the guide:
https://github.com/opentx/opentx/wiki/Translators-Github-Guide

Datospel, 2016-01-09:
thank you, I'll try to manage this. Don't close this issue, please, in case if I'd have some questions. Than you

kilrah, 2016-01-09:
There was already a stub of a Russian translation in companion, that has been disabled because only a fraction of entries were present. See the 2 commits at the end of https://github.com/opentx/opentx/issues/712 to reenable.

Datospel, 2016-01-09:
ok. I'll check it

closed 2018-06-06 by mpaperno