opentx/opentx #3219: Impossible to change anything using Open TX Companion 2.17 with my new 9XE

G2loq2, 2016-01-09:
https://youtu.be/AXLPpe92iQE
 As you will see in the video, it is impossible for me to change anything using Open TX Companion 2.17 with my new 9XE When I start my TX after any change, even the name of the model, it said Bad EEPROM Data, it reformat everything and I loose all my information. Why do I do wrong ?

projectkk2glider, 2016-01-09:
Judging from your bootloader version you have FrSky version of opentx which has version number 2.1.4. They apparently made some changes that makes it incompatible with companion. Flash latest opentx and you should be good.

closed 2016-01-09 by projectkk2glider

kilrah, 2016-01-10:
You have companion set for a taranis plus instead of a taranis X9E.