opentx/opentx #3221: Different naming of input controls in wizard

fraternl, 2016-01-10:
Yesterday I had to explain how a Taranis should be programmed to two experienced flyers.
They had a lot of difficulty in trying to understand the concept and after they understood it they somehow questioned it was logical.

The most confusing were the terms used to describe the names of the inputs/mixers.

As an example they asked me how I would program some elevator compensation when using the rudder.
The moment I told them that I had to add a mix to the elevator channel they started to lose me.
They were inclined to add something to the rudder mix.

I believe a lot of confusion is already made after the LUA-wizard creates a default model and names the inputs to throttle/rudder/elevator/aileron
This is of course a legacy way of describing them, but shouldn't they really be
Thrust/yaw/pitch/roll?

I can easily change that in the LUA-script myself of course.  

Do you agree with me that the inputs should really be named after the actions instead of the controls on the airplane?
At the same time it would make a more pronounced difference between sticks and inputs.

If you agree (I don't know this yet), do you think this should warrant a change?

jtaylor2, 2016-01-10:
>This is of course a legacy way of describing them, but shouldn't they really be
Thrust/yaw/pitch/roll?

Legacy is important.  Almost every radio from the beginning has called the airplane setup by the aircraft controls (throttle/rudder/elevator/aileron).  People are used to it and understand it.  Even the people using the radio for cars, boats and other devices are used to it and understand it.  And your suggestion is no more accurate in many cases than the existing names.  For example when you program  a flap function to ailerons you are still programming the ailerons, but (hopefully) not affecting roll.

Probably changing the names for the helicopter and multi-rotor wizards makes sense, but I think changing it for airplanes would just create confusion.

kilrah, 2016-01-10:
Frankly all the radios I know work on exactly that same principle, using the Thr/Ail/Ele/Rud as function names. 
The only thing some could be confused with is not to notice the difference between the raw stick and the similarly named processed input, but if anything it's the stick name you'd want to change. But then it's a generic J1-J4 à la Futaba which is just as annoying because you never know which is which.

fraternl, 2016-01-10:
@Kilrah, I agree with you concerning the name of the stick that should be changed, but at the same time I don't want to change OpenTX itself and respect unspoken "legacy rules".

Because other radio's have a different concept they have no other choice than naming that stick/input "rudder/elevator/aileron"

When programming a flap to an aileron it's still the "aileron" in the mixer page (which IMHO should still be named aileron) that will get actuated by an input that's called "flaps" which in turn could be coupled to the stick named "throttle".

fraternl, 2016-01-10:
@jtaylor what I meant to say is that your example with the flaps is exactly what I mean.

In my setup I would name it like this (no change in openTX required, only the LUA-wizard)

Inputs

Flaps.  Thr 100
Roll.      Ail 100

Mix

Ail L
Roll 100
Flaps 50

Ail R
Roll -100
Flaps 50

kilrah, 2016-01-10:
> When programming a flap to an aileron it's still the "aileron" in the mixer page (which IMHO should still be named aileron) that will get actuated by an input that's called "flaps" which in turn could be coupled to the stick named "throttle".

That's exatly what you can/should do in OpenTX too.

- Name your output channel 1 "aileron" on the outputs page
- Create an Aileron input controlled by the Ail stick (it's there by default)
- Create a Flaps input controlled by the Thr stick
- In the mixer on the "aileron" channel (CH1) mix in the Aileron input and the Flaps input.

It's precisely the same as everybody else, even if the actual way to map and assign things is more explicit than most who would either do some of it "under the hood" without exposing it, or expose it in a place where you have to go look for it.

fraternl, 2016-01-10:
Yes, I know I can do this and OpenTX itself makes it entirely possible to do it as I want it.

But users build their model upon the result of the LUA-script and I'm thinking that the slight change in making the inputs emphasize the difference between openTX and the other menu-driven concepts and might (maybe not true) result in a better understanding of openTX.

I completely understand if you prefer to keep things as they are. I think I will probably change the LUA-script for my own purpose, but am not even sure about that either.

The reason I made this ticket is to find out if others feel like I feel and if they can make me change my mind or the other way around

kilrah, 2016-01-10:
If anything you could improve the script to implement things properly this way, with the channel names and everything where it should be. As it is it was made as a quick and simple entry point but without too much effort on making it an example to follow.

fraternl, 2016-01-10:
I would gladly do this. To be honest, these are merely some minor changes.

But does it mean you agree with my reasoning? And even if so.... It's a psychological thing, too...

I don't want to impose my personal views on a community without a majority agreeing with me.

kilrah, 2016-01-10:
Well I am personally OK to improve the wizard and add channel names for example, but not to change the input names ;)

fraternl, 2016-01-10:
Am I correct you want it like this?

Inputs
Thr  Thr 100
Ail Ail 100

Mixes
CH01 Ail L
iAil 100  Roll
iThr 50 Flaps

CH02 Ail R
iAil -100  Roll
iThr 50 Flaps

I was more thinking of:

Inputs
Flap Thr 100
Roll Ail 100

Mixes
CH01 Ail L
iRoll 100
iFlap 50

CH02 Ail R
iRoll -100
iFlap 50

kilrah, 2016-01-11:
> Inputs
Flap Thr 100
Roll Ail 100

That's what we say nobody has ever used and is thus will likely not understand. 

II was only thinking of adding the control surface names to the channels on the Outputs screen (LAil, Rail, Elev, Rudd etc)

closed 2016-11-27 by fraternl