opentx/opentx #3223: Radio/companion LUA 2.1.7 some value not accessible

pointu, 2016-01-10:
since 2.1.7 getValue() return nothing for clock, tx-voltage timer1 timer2 ... certainly some other can't verify all.

what's the matter ?

projectkk2glider, 2016-01-10:
Confirmed, its a mistake I made in Makefile.

projectkk2glider, 2016-01-10:
Missing values:
```
> clock                    RTC clock [minutes from midnight]
> gvar6                    Global variable 6
> gvar7                    Global variable 7
> gvar8                    Global variable 8
> gvar9                    Global variable 9
> timer1                   Timer 1 value [seconds]
> timer2                   Timer 2 value [seconds]
> timer3                   Timer 3 value [seconds]
> tx-voltage               Transmitter battery voltage [volts]
```

closed 2016-01-10 by projectkk2glider

pointu, 2016-01-22:
Question : with this lua is not usable for use or testing, why dont you do a makefile again with a 2.1.7.1 version (why not) because for common people that's not useful to do, and there is a lot of new things in this version.
Thanks

kilrah, 2016-01-22:
It will be fixed in 2.1.8 when it is released.

pointu, 2016-01-22:
Of course, but this version is not really usable like that, that's why i told you.

kilrah, 2016-01-23:
It is usable for all other functions, like flying models. 
A release is a lot of work both for us and for users, so we don't release a new version every time a little thing is fixed or added...

You can build OpenTX from source yourself or use the nightly builds if you want to move on faster.

commit 5cb3ef82 (2016-01-10): Fixes #3223: some lua fields were not exported (CPUARM)
commit 8780535f (2016-01-10): Fixes #3223: some lua fields were not exported (CPUARM)