opentx/opentx #3224: Support for Teensy (3.2) board?

geekness, 2016-01-11:
I was wondering if there was any thought into porting OpenTx to the Teensy board?
https://www.pjrc.com/teensy/teensy31.html

I know there is support for the DIY mega 2560, but I thought the Teensy is a big upgrade from that.
Also, there could be an opportunity to make a windows GUI as the interface.

kilrah, 2016-01-11:
The mega2560 radio exists because that processor is nearly directly compatible with the one in the 9X.
The teensy has nothing in common with any platform we currently support whether in hardware or in anything related to user interface, so a very strong motivation would be needed to undertake such a large amount of work. 

If you're looking for a higher performance DIY board than the 2560 you might want to look at the STM32F4discovery board. That has the advantage of using the same architecture as the Taranis family (closest to the X9E).

closed 2016-01-11 by kilrah

geekness, 2016-01-11:
Has there been any thought or movement in porting OpenTx to the STM32F4disco?

kilrah, 2016-01-11:
No, but it should be a simple enough project for someone who would want to add support for it, very similar to what Franck did for the MEGA2560.

UltraWhiteAngel, 2021-01-12:
Any Links would be greatly helpful.
If it is possible, 

> Has there been any thought or movement in porting OpenTx to the STM32F4disco?

> No, but it should be a simple enough project for someone who would want to add support for it, very similar to what Franck did for the MEGA2560.

So far I have this minor speculation that using a STM32F4 makes more sense than a Mega2560,  Are you aware if it has already been done?