opentx/opentx #3226: MEGA 2560 PPM frame encoding

Gabapentin, 2016-01-11:
Hi to all and thank you in advance for your kindly cooperation

I'm trying to build a radio with Arduino MEGA 2560 project

everything seems to be working but I think there's an issue about PPM frame format

I use TX with Flysky protocol and 6/8 channels receivers

This is a PPM frame capture of MEGA 2560 (6CH)

[image]

Seems to be long delay on 1st channel start, 
the receiver mapping channel (1) to other channel (5)

I've tested several radio projects with no issue with PPM encoding or with receiver

This is a PPM frame capture (6CH) from other PPM encoders without receiver problems

[image]

thanks a lot for our effort about opentx project

projectkk2glider, 2016-01-11:
Isn't this the same problem as #2938 for which the solution is in #2981?

kilrah, 2016-01-11:
Yep.

closed 2016-01-11 by kilrah

Gabapentin, 2016-01-11:
Thank you for your reply Damjan

Yes I know is the same problem as #2938 
<https://github.com/opentx/opentx/issues/2938>   and solution is in 
#2981 <https://github.com/opentx/opentx/pull/2981>
but I'm totally newbie about firmware development and sorry for my 
incompetence

on linux os
I've tried to compile several times with pulses_avr.cpp update file
copied from slope soarer github (is correct procedure or not)
but I have no luck... I believe to be wrong in something....

PPM frame remains the same even when I've loaded hex file generated from 
compiler

You have a little suggest

Please do not send me to hell

Thank you very much for your collaboration

Best Regards Patrick

kilrah, 2016-01-12:
If you want to do this and build yourself you must be sure to take the file from sloped_soarer's `issue_2938_MEGA2560-the-first-PPM-pulse-in-the-frame-longer` branch, not the default `master`.

[image]

Gabapentin, 2016-01-12:
> clipboard01

Thanks a lot Andre

I have to eat humble pie

I'vefollowed your suggestion

Now working fine

owe you a beer

Merci encore

BR's Patrick