opentx/opentx #3229: Addition of MIN to source list in Mixes Screen

Oldgibber, 2016-01-13:
The MAX source is quite useful in the Mixes source list, especially when programming switches. It would seem logical to also include a MIN as well, especially for those whose logic (or maths) is not that hot. 

It just seems odd to have to use the line MAX (Weight -100%)   ie multiply max value by minimum weight
when one could use MIN (weight 100%)  ie multiply min value by maximum weight

projectkk2glider, 2016-01-13:
I think it is not needed.

mhotar, 2016-01-13:
I would see the MAX renamed do FIX and the field description "weight" changed to "value" for this case when FIX is selected. But we are used to use MAX already.

lshems, 2016-01-14:
I need it for programming easy lists for changing a source.

het volgende:

elecpower, 2016-01-14:
MAX and MIN would be just as obscure as each other for newbies. An alternative suggestion is to use the word SET.

kilrah, 2016-01-14:
I voted for FIX a few years ago - but IMO no need to change anything people are used to. I'd however prefer that to adding a MIN.

lshems, 2016-01-14:
I stil don't think people have understood a very interesting use for min.

Lots of people us inputs to make programs understandable and maintainable
by assigning a source to that input, and use the input as reference to that
source.
In that way you can for instance very easy change a three state flap switch
to a flap slider, by using an input in a mix, and change the source of that
input from SA to LS for instance.

If you have lots of these inputs, sometimes you have to make choices,
because you don't have enough control inputs for all possible inputs. For
instance you have used all physical switches, but you want yet just another
feature to be controlled by a physical switch.

So, you decide to sacrifice another feature, making it a fixed choice to
free that physical switch for the new feature.

You do that by assigning MAX as a source, so the input will get a fixed
value. This will fix this feature in the state corresponding to the down
state of the physical switch it had as a source before.

If you want the feature, controlled by the input that has now become fixed,
to remain in the state of the original switch in up position, you have to
assign -100% as a weight.
If you want the feature to act as if the now removed physical switch was in
the middle position, then you need to assign weight 0%, or source -- to
that input.

Point in all this, is that it is not intuitive.
If we had a MIN value, in the list behind the MAX value, then we could
simply scroll through the source list and assign whatever fixed 'switch'
replacement value we want, and all in one go, being MIN for switch up, MAX
for switch down, and -- for switch middle.

It would help any choice you want to offer users of a lua script to change
an input also much easier to program, because you don't have to check if
MAX has been selected to decide if you also need to change the weight from
100% to -100%.

I really hope you could try and see the usefulness, its more powerful than
you think at first glance!!!

Oldgibber, 2016-01-14:
Thanks Ishems, that has put exactly my point very elegantly. A MIN value would make setting up switches much easier. 

Can I also point out that, certainly here in the UK, the majority of modellers join modelling not in their youth but when they near retirement age or when we retire. My own large club certainly reflects this. our generation of over 65s like most round the world had no computers in our youth, and no curriculum which included coding or logic. 

OpenTX is a real gamechanger and extremely powerful, however,sometimes one needs to stand back and see it from a beginner's perspective. Which is easier to explain and which looks more elegant:
 
MAX Weight(-100%) Switch(SA↓) 
MAX Weight(+100%) Switch(SA↑)

or 

MIN Weight(+100%) Switch(SA↓) 
MAX Weight(+100%) Switch(SA↑)

Perhaps too one ought to consider the way models are changing. In the last few years more and more foamies are coming on the market which are very inexpensive and include such things as lights and landing gear. Thus the use of switches to control these are becoming much more of an essential feature, and hence the plethora of switches on new transmitters.

kilrah, 2016-01-14:
> If you want the feature, controlled by the input that has now become fixed, to remain in the state of the original switch in up position, you have to assign -100% as a weight. If you want the [...]
If we had a MIN value, in the list behind the MAX value, then we could simply scroll through the source list and assign whatever fixed 'switch' replacement value we want, and all in one go, being MIN for switch up, MAX for switch down, and -- for switch middle.

Sorry, but to me learning/remembering that MIN / MAX correspond to switch up/down is exactly the same as learning/remembering that -100%/+100% correspond to switch up/down. Only that learning the former will be useful to in more situations, and learning the latter might trigger some assupmtions in the unsuspecting user such as MIN always being MIN - which is wrong, the output settings and servo reverse are still in effect.

The current weight approach favors learning and getting used to the OpenTX control flow by helping the user think of the entire chain of settings. 

So still for just FIX as MAX is not a good description of what it is.

elecpower, 2016-01-14:
Agree with @kilrah that MIN & MAX should not correspond to a switch up/down due to other OpenTX flexibility. 
In my vote for FIX or similar, both MIN & MAX can give the impression to some that the value is bounded at one end by the MIN or MAX when for the purpose of calculations, as I understand, it is a fixed value. Other calculations and settings may clip, invert, modify, etc to achieve the final value sent to say a servo.

tomkfly, 2016-01-14:
I believe  MIN is an unessasary complication.

Oldgibber, 2016-01-14:
Elecpower, surely not. 
If we just had the line 
**MAX  Switch(SA↓)** then yes that would give the impression that MAX was a fixed value that would be sent to the receiver under all conditions, but
**MAX Weight(-100%) Switch(SA↓)** says "take the maximum value, ie 100, and multiply it by the weight. Thus the impression is immediately made that MAX is manipulated by further conditions.

The use of FIX as in **FIX Value(-100%) Switch(SA↓)** certainly gives the impression that -100% is a value that will be sent to the receiver under all conditions.

tomkfly, 2016-01-14:
Agree, fix does seem like an unchangeable value.

kilrah, 2016-01-14:
Then SET would too, especially given that people know about the "Override" SF. CST for "constant"?

elecpower, 2016-01-14:
@kilrah you have hit the problem on the head with CST. Just pass the value as is (the definition of a constant) to the computation engine. What the engine does with the value is determined by the underlying logic.

Oldgibber, 2016-04-16:
Had a thought about this. Seems general agreement that MAX is not ideal, but having MAX/MIN or FIX or SET are not ideal. How about **USE**?

USE Weight(+100%) Switch(SA↓)
USE Weight(+100%) Switch(SA↑)

ie
use a weighting of +100% when switch SA↓
use a weighting of -100% when switch SA↑

It would appear to be a very simple programming change.

mhotar, 2016-04-16:
Another option can be to split SOURCE to SOURCE TYPE and CONTENT.
TYPES can be: inputs, raw stick -100/100,raw stick 0/100, fixed value, global variable, switches, logical switches, channels, trainers ....
We already have these group types with icons, so we can add them as a fixed line in the inputs/mixes.
This can go probably more deeply ..... no more time right now

Oldgibber, 2016-04-16:
Isn't that simply making things a good deal more complicated for no real gain at this stage?

closed 2016-04-27 by Oldgibber