opentx/opentx #3230: Taranis Plus 2.0.18

urbans90, 2016-01-13:
Hello. When the 2.0.19? And who can make a firmware fix for 2.0.18? A bug in the folder when scrolling up. Flashing when the backlight. Or tell me how to fix it all by hand. Thank you.

projectkk2glider, 2016-01-13:
Please describe your issue more clearly, I cannot make anything from your post.

urbans90, 2016-01-13:
Here's a video like when paging up all is lost. And there's also a flash of blue when turned on. https://goo.gl/photos/mYo5zr5Awh6uDpKZ7

projectkk2glider, 2016-01-13:
Did it work in 2.1.7? Does it happen only in directories with a lot of files or in all directories?

urbans90, 2016-01-13:
I did not set 2.1.7. This firmware 2.0.18. This occurs when flipping have not depending on the quantity. And only when paging up from any position.

kilrah, 2016-01-14:
How many files do you have in that "en" folder? Looks like way too many.

urbans90, 2016-01-14:
If you put the firmware 2.0.9.2 is no such. And it's not a lot of files.

projectkk2glider, 2016-01-14:
Confirmed on simu, and 2.1 is not affected.

urbans90, 2016-01-14:
How to fix it? What program? How to rebuild the firmware?

projectkk2glider, 2016-01-15:
@urbans90 wait for 2.0.19 release...

projectkk2glider, 2016-01-15:
A one character bug :relaxed:

closed 2016-01-15 by projectkk2glider

projectkk2glider, 2016-01-15:
Tested on my Taranis, it works.

projectkk2glider, 2016-01-16:
@urbans90 did you test  2.0.19, is it OK for you now?

urbans90, 2016-01-16:
Yes, everything is great.

urbans90, 2016-01-16:
And you can somehow swap adjust the illumination color. To the right was the blue and white on the left? And then when when you turn if you choose something white is a flash blue, then lights white lights. On my video which is higher than it should be seen. The firmware 2.0.9.2 was white lights on the left in the settings. I apologize for my bad English.

projectkk2glider, 2016-01-16:
The backlight gets turned on before the radio settings are read, therefore the user backlight setting is unknown at that time. 

So a default preset is used for a moment, which I think it is better than to leave it off and turn it on after the settings are known.

urbans90, 2016-01-16:
Good. Let then so be it.

commit fd2e0e49 (2016-01-15): Fixes #3230: SD manager bug when scrolling up fixed