opentx/opentx #3231: Create a SCRIPTS/LIBRARY folder by default

jcowgar, 2016-01-14:
In lua, we can use dofile() to load reusable code. It would be nice to have a place to install reusable code without it appearing in the scripts menus. For example, some reusable code for a TELEMETRY script. It needs some place to live.

My immediate use for this is a new library I began creating, opentx-ui for use with my opentx-laptimer. opentx-ui contains menu and configuration file code. http://github.com/jcowgar/opentx-ui and http://github.com/jcowgar/opentx-laptimer for examples.

lshems, 2016-01-14:
Check my blog on rcgroups. I created it simply for myself. No need for a
folder to be created by default??

het volgende:

jcowgar, 2016-01-14:
Yes, that is what I did as well, but I figured if it existed by default, as other folders do, that it would then create a standard amongst Lua programmers on where to put such files. So one person doesn't depend on /SCRIPTS/LIB, another, /SCRIPTS/SUPPORT, another /LIBRARY, etc...

jcowgar, 2016-01-14:
What is your blog on RCGroups? Can't find your username.

lshems, 2016-01-14:
L shems

het volgende:

closed 2019-11-15 by stale[bot]