opentx/opentx #3232: Unable to read replayed cell values

hymanroth, 2016-01-15:
Version 2.1.17 / Ubuntu 12.04 (64 bit)

I'm playing back a log file with FLVSS data and can't read the values in a lua script.

I've tried getFieldInfo("Cels") and getFieldInfo("Cel") but in both cases the call returns null.

I can't work out whether I'm doing something wrong or I've found a bug.

Happy to test more if required.

projectkk2glider, 2016-01-15:
You should use the actual name of your cell sensor for the function call. The data format returned is described in https://opentx.gitbooks.io/opentx-lua-reference-guide/content/general/getValue.html The return value in this case is a table.

hymanroth, 2016-01-15:
Thanks for your comment. Actually I did refer to that that document whilst writing the script, but it appears to me that all the fields listed refer to local values on the transmitter.

projectkk2glider, 2016-01-16:
The telemetry fields are referred by their actual names, which are not fixed, they are user definable so they can not be in Lua fields list document anymore.

projectkk2glider, 2016-01-16:
Another possibility is that telemetry simulator does not support the playback of Cell sensor. @dsbeach  should know more.

hymanroth, 2016-01-16:
Thanks, it's clear to me now. I tried playing back the logfile whilst trying to autodiscover the FLVSS sensor but it did not work. I imagine it would work on the radio, but this will make Lua coding with sensor data incredibly tedious if the data cannot be emulated in Companion.

dsbeach, 2016-01-16:
You should be able to playback FLVSS data. Let me see if I can help, or if there is a problem that needs addressing.

The method that works best is to discover the sensor in the transmitter, then copy that model to companion. Load the model in companion, and make sure it is fully initialized before opening the telemetry sim. Once the telemetry sim window is open it should show the correct instance ID and the log playback should work fine.

If you still have issues you can email me an eepe and log file (david.s.beach@gmail.com) and I'll see what is going on.

David

hymanroth, 2016-01-16:
Thanks David. I managed to read the values in Lua by using the telemetry sim. When I get the time I'll set up the physical model and do as you say. I'll get back to you if there any problems.

dsbeach, 2016-01-16:
Sounds good.  I'll go ahead and close the problem.  A lot of the confusion comes from being able to discover the sensors using companion, but not being able to save them when they are discovered via telemetry simulation.

What is really needed is to be able to add the custom sensors (by sensor type) with in the "Edit Model" function of Companion. Unfortunately that has not yet made it to the top of anyone's 'to do' list.

closed 2016-01-16 by dsbeach

projectkk2glider, 2016-01-16:
You might also have a problem with sensor instance, for which there will be a fix in 2.1.8, see #3207.