opentx/opentx #3233: 2.1.7 - Jitter on Channel Monitor and receiver output.

Rob2160, 2016-01-16:
Can we re visit this please. 

I understand the issue was closed (2774) but both of my Taranis are still experiencing jitter with 2.1.7. 

The jitter is present in the Channel Monitor, and is also visible in the receiver output.     

My radios are:  1 x Taranis B01 and 1 x Taranis Plus B02.

Both radios are jitter free on all 2.0x version firmware.

Video 1 - Jitter with 2.1.7 (Taranis Plus)

https://www.youtube.com/watch?v=7Hv4GdGJasM

Video 2 - No Jitter with 2.017 (Same Taranis Plus - same day) 

https://www.youtube.com/watch?v=l0ynqLULJE0

Video 3 - Jitter from TX is visible in Receiver output 

https://www.youtube.com/watch?v=4v9-JSiZwt4

Video 4 - Jitter still present with Internal and External RF turned off. (no external module fitted here)

https://www.youtube.com/watch?v=SQTjbXBN-gw

I can replicate these results every time, on two separate radios that both have zero jitter with 2.0X version firmware.

projectkk2glider, 2016-01-16:
@Rob2160:
* Tell us what firmware options you are using. 
* What are you seeing on ANALOG INPUTS page (radio setup, page 6/8 on OpenTX 2.0.x)? 
* Try using a simple default model (4 channels) is there any change in jitter?
* Try the same simple model, but this time remove trim from all mixes, any change?

Here is my Taranis B on 2.0.19:
* default 4 ch model
* jitter around +-0.1 percent (or even less) on CHANNELS MONITOR , I get more jitter just after power-up, but then it stabilizes after several minutes.
* dead stable readings on ANALOG INPUTS page

projectkk2glider, 2016-01-16:
Here is my Taranis B on unreleased 2.1.8:
* same default 4 ch model (automatically converted on firmware upgrade)
* visible small jitter on on ANALOG INPUTS page (max jitter is 1)
* hard to tell if jitter on CHANNELS MONITOR is more, but it could be.

So there is some difference, but it is small on my radio. I think we must have changed something in ADC conversion to have more jitter in ANALOG INPUTS.

Rob2160, 2016-01-16:
Thanks for the reply. 
Do you mean build options when downloading the firmware?  If so, everything is unselected as per the image below. 

Taranis Plus B02 on 2.1.7
I created a default 4 ch model:
 - Visible Jitter on Analogue inputs page
 - Visible Jitter on Channels Monitor page
No change with trims enabled or disabled. 

Then switched to 2.0.19 on same radio

projectkk2glider, 2016-01-16:
That is a very good report, thanks. I think we have all information that we need. 

But I must say right of the bat that the jitter is not so big to be very concerned. But it will be interesting to find out which change is causing it. I will take a look...

projectkk2glider, 2016-01-16:
Here are my findings after comparing sources, making test builds, etc...

The jitter is present on both version 2.0 and 2.1, it's just less visible on 2.0. Perhaps there is more jitter on 2.1 but that is not caused by any change in ADC processing code (since there weren't any changes there that would affect this).

The 2.1 has main thread that runs LCD display running differently from the 2.0, so the refresh rate is different. That alone could account for visible difference. Next possibility is that something else is running on CPU when the ADC acquisition is running and it generates more noise on 2.1. 

@Rob2160 could you do another test to prove my theory. Load 2.0 and connect it to your Cleanflight board. In you video on 2.1 the jitter shown by the Cleanflight is only 1 micro second (which is so small that it does not affect anything in reality). I think that you will see the same jitter on 2.0. The point when you see no jitter when you turn of the radio can also be explained - your receiver is going into failsafe and sending out the last value.

You can also use a mixer setting like this to increase jitter 8 times on CH4:
```
CH01     [I1]Thr Weight(+100%)
CH02     CH1 Weight(+200%)
CH03     CH2 Weight(+200%)
CH04     CH3 Weight(+200%)
```
With that setup I can see jitter on my Taranis on both 2.0 and 2.1.

483917, 2016-01-16:
Hello i have also noticed this jitter on my
Taranis B with firmware: "opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.7N607.bin"
could this jitter be related to the #3178 Interrupt Bug ?? mabye ??
just a thought

Rob2160, 2016-01-17:
Hi projectkk2glider,

I have made two additional videos - the first is comparing the jitter in the Cleanflight Channel sliders.   There is a noticeable difference between 2.1.7 and  2.0.19.

https://www.youtube.com/watch?v=dTSh_oKcHko

The next video I compare the jitter by using your amplification mixer - master channel is CH1 (Throttle stick and the most amplified is CH4

The difference in jitter on CH 4 between 2.0 and 2.1 is now huge and also reflected in the Cleanflight channel sliders. 

https://www.youtube.com/watch?v=LGiaHB0tFl8

projectkk2glider, 2016-01-17:
>could this jitter be related to the #3178 Interrupt Bug ?

I doubt it.

@Rob2160 Thanks again for great videos. You again confirmed that jitter is bigger on 2.1. I will look at this some more.

marcosrullan, 2016-01-17:
Hi, I have similar problem with my taranis plus. I flashed with 2.0.x and 2.1.7 and i saw the same problem. My taranis is from 19/08/2015.

projectkk2glider, 2016-01-17:
I've made a test firmware that measures and displays jitter. The measured values are displayed in ANALOG INPUTS screen:
[image]

I have added two  columns between raw value 0000 and the actual channel value, for example:
`A1:0000    0      0      0` = `A1:<raw value in hex> <raw jitter> <average jitter> <stick value in %>`

* `<raw jitter>` is maximum difference in the ADC reading in the last second
* `<average jitter>` is maximum difference in the filtered ADC reading (this one is used for mixes/inputs) in the last second. This is the important value to watch. It gives us the information how much jitter is on given analog input. 

With this firmware, on my Taranis I get:
 * on 2.0:
  * average values of `<raw jitter>` and  `<average jitter>`  (when looking at A1, A3 and A5 values): **22, 3**

 * on 2.1
  * average values of `<raw jitter>` and  `<average jitter>`  (when looking at A1, A3 and A5 values): **25, 4**
  * less jitter with backlight off

* turning off internal RF module or using different mode (D16, D8) makes no difference to jitter on both version. 
* backlight is affecting jitter (more jitter with backlight on) but only in 2.1 version.

Videos:
 * 2.0 https://www.youtube.com/watch?v=tfJx1CVhr5w
 * 2.1 https://www.youtube.com/watch?v=9BP6rDDvUwk

It will be interesting to see what other radios are doing, mine for example has jitter on both versions, just less of it on 2.0 (3 vs. 5).

Here are the firmwares for testing this issue. Please do not fly with these firmwares and make a backup of your settings before flashing them:
 * for Taranis: 
  * http://downloads-21.open-tx.org/test/opentx-20-jitter-taranis.bin 
  * http://downloads-21.open-tx.org/test/opentx-21-jitter-taranis.bin
 * for Taranis Plus: 
  * http://downloads-21.open-tx.org/test/opentx-20-jitter-taranisplus.bin
  * http://downloads-21.open-tx.org/test/opentx-21-jitter-taranisplus.bin

projectkk2glider, 2016-01-17:
Before I implemented jitter measurement I tested:
 * disabled interrupts while ADC/DMA was running - no visible difference
 * increased ADC sample time - no visible difference

483917, 2016-01-17:
Ok did the firmware test on my Taranis B01

on 2.0:
average values of <raw jitter> and <average jitter> (when looking at A1, A3 and A5 values):
A1: 30 ,5
A3: 29 ,4
A5: 29 ,4

on 2.1
average values of <raw jitter> and <average jitter> (when looking at A1, A3 and A5 values):
A1: 31 ,7
A3: 30 ,8
A5: 31 ,8

all tests with backlight ON

bsongis, 2016-01-17:
It makes me think that we will have a problem with the Horus as well. Even calibration of the 6-pos switch is a problem, it detects more positions when the other ADC move ...

kilrah, 2016-01-17:
>it detects more positions when the other ADC move ...

Then that would suggest the same kind of cross-talk some have reported on X9E (there's an issue somewhere). 

>increased ADC sample time - no visible difference

What exactly did you increase and by how many?

Rob2160, 2016-01-17:
With my Taranis Plus

on 2.0:
average values of and (when looking at A1, A3 and A5 values):
A1: 27 ,7
A3: 26 ,6
A5: 24 ,5

Video: https://www.youtube.com/watch?v=Jp2Ewu1zRkU

One thing of note - with the test 2.0 firmware I was noticing the same level of jitter in the Channel Monitor as 2.1.7 and 2.1 test firmware.   (visible in video) - there was no jitter in the channel monitor with the normal 2.0.19

on 2.1
average values of and (when looking at A1, A3 and A5 values):
A1: 33 ,8
A3: 32 ,7
A5: 32 ,7

Video https://www.youtube.com/watch?v=5awAGHkzzX8

I just flashed back to 2.0.19

projectkk2glider, 2016-01-17:
>What exactly did you increase and by how many?

I changed `SAMPTIME` from 2 to 3.

>One thing of note - with the test 2.0 firmware I was noticing the same level of jitter in the Channel Monitor as 2.1.7 and 2.1 test firmware. (visible in video) - there was no jitter in the channel monitor with the normal 2.0.19

I find this very interesting. This might be the indication of something deeper. The fact is that my 2.0 test version did not change anything significant. In fact when the jitter values are not shown they are not computed either. The changes are https://github.com/opentx/opentx/compare/cf0e7c4a441eb04744d466c3cc9eed48874d9bdf...951b316656c3dc7ad33a0439dfbdac2bbae4f54c The only thing changed is the time it takes for the `getADC()` code to run.

projectkk2glider, 2016-01-17:
Here are some more of my ramblings.

We can observe from the data above that `<raw jitter>` is around 30 on all radios. What is exactly that value showing? 

Here is how sampling is done:
 * every 2 - 3 ms or so (depending on mixer complexity and other tasks) we take a reading of all analog inputs.
   * all analog inputs are read one by one in a rapid succession, this process takes estimated time of 20us.
   * the read is repeated four times and measured values are added together (separately for each analog input) and divided by 8. This is the value that is used for further processing. This value is also used for  `<average jitter>` calculation.

So in effect (looking at one analog input only) the time-line is like this:
 * four samples in rapid succession (spaced roughly 20us)
 * 2-3 ms of not measuring anyting
 * repeat

In effect we are only sampling for 4 * 20us every 2ms, which is only 1/25 of the time. 

So if we come back to `<raw jitter>` value of 30. This means that over the one second period the maximum and minimum  value seen for particular analog input has varied by that amount. The 30 is difference between max and min values.

This tells me that baseline noise on input is 30 units (out of 12 bit ADC range of 4096) which is 0.7%. 

The nature of this noise (mostly frequency) will affect how our averaging/filtering algorithm is working. Since we only average for a short time (remember four readings spaced 20us) we only filter for high frequency noise. The low frequency noise pases trough and is seen as  `<average jitter>` value.

Any small change in the (2-3ms) time between the readings can affect the output, because it is changing the position of sampling period (remember we measure only 1/25 of the time). 

My theory is: This variance in sampling time can affect the jitter. That is why the jitter on 2.0 with my version of firmware is bigger than on stock 2.0 firmware.

Anyhow I think a better method of ADC measuring can be done, now that we can measure jitter it should be easy to measure its performance. I will try a couple of ideas....

projectkk2glider, 2016-01-17:
Here is my first idea https://github.com/opentx/opentx/commit/faaa14f71dc4572666a1383d17a6664e0d7429e0, the jitter filter. Works like this:
 * after the new round of ADC reads is done compare the new averaged value with the old one and compute `absolute difference`
 * if `absolute difference` is bigger than 10, then use the new value directly (this is how it worked all the time before)
 * if `absolute difference` is less than 10, then use a heavy (slow) filter to compute new value.

With this setup the jitter is now in a range of 0-1 on my radio. Pretty good I think.

Rob2160, 2016-01-18:
The jitter filter firmware works on my radio - I also get jitter 0-1.

A short video showing the Channel Monitor (std 4 channel model and also the 8 x amplified mix) and the results in Cleanflight. 

https://www.youtube.com/watch?v=uOPdTk28mrE

483917, 2016-01-18:
I tested the "opentx-21-jitter-filter-taranis.bin" on my Taranis B01
and i also get jitter 0-1 = much better :+1:

projectkk2glider, 2016-01-18:
Thanks for reporting, I think it is a good way to get rid of the jitter. 

I will let other developers look at this and then we can include it in the mainline branch. Perhaps we can use it as compile time option or even runtime radio setting.

elecpower, 2016-01-18:
Great work to you all on this one. I assume a fix will improve the radio calibration and eliminate centre position warnings on S1, S2, LS & RS plus clean up telemetry. Benefits all round.
The radio setting seems a good idea in that the jitter from testing is inconsistent and over time the sensitivity especially with different hardware will neccessitiate tweaking. A setting would eliminate the need to keep revisiting the code to alter the value. The inital value may even be possible as part of calibration.

projectkk2glider, 2016-01-18:
There is currently no oversampling and jitter filtering done when the calibration is running. I wonder why. @bsongis is this the best solution, wouldn't the calibration also benefit from the oversampling and jitter filtering?

closed 2016-01-18 by bsongis

LapinFou, 2016-06-24:
FYI I read something which may interest you: http://www.rcgroups.com/forums/showpost.php?p=35100721&postcount=48613

LapinFou, 2016-06-24:
Just in case, here is the interesting part:

> Regarding the jitter, I wonder if the analog to digital operation for reading the sticks has been changed. OpenTx was forked from er9x. This included an option on how the sticks were read, SING (single read) OSMP (over sample where each was read several times and the average used) and FILT (where a more complex filtering algorithm was used).
> I note this option setting appears to no longer exist in openTx.
> It looks to me that 2.0.19 may be using OSMP, but 2.1.x is using SING.
> In ersky9x (which also runs on the Taranis) I have retained the option. Switching between SING and OSMP, I get a similar change in the jitter level.
> Ersky9x also has options for setting a "deadband" for the centre position of the sticks that may be used to remove jitter at the central position without affecting the resolution or response to movement.

Rob2160, 2016-06-24:
Excellent thanks for the heads up Sebastien.

PR #3240 (2016-01-18): Fixes #3233: Added ADC jitter measurement (use MEASURE_JITTER=YES to …
…enable), Added ADC jitter filter (use JITTER_FILTER=No to disable)

So jitter filter is enabled by default, while the measurement in disabled.

Closes #3233
commit 30a7de6f (2016-01-18): Fixes #3233: Added ADC jitter measurement (use MEASURE_JITTER=YES to enable), Added ADC jitter filter (use JITTER_FILTER=No to disable)
commit 3e90ab72 (2016-01-18): Merge pull request #3240 from opentx/projectkk2glider/issue_3233_adc_jitter_filter_and_measure

Fixes #3233: Added ADC jitter measurement (use MEASURE_JITTER=YES to …
commit 4aa6bd12 (2016-01-19): Fixes #3233: Added ADC jitter measurement (use MEASURE_JITTER=YES to enable), Added ADC jitter filter (use JITTER_FILTER=No to disable) (ported from master)